OpenEPaperLink/OpenEPaperLink #350: AP resets after sending malformed POST request.

mukowman, 2024-07-11:
Testing out posting to save_cfg
I found if I miss some of the parameters like "alias" instead of returning a 400 error, the AP resets.

LeftPinkie, 2024-07-15:
What is the env and version of your AP? What is the HTTP status code that you're getting. For example:
```
$ curl --json '{"mac":"0000021F7E3A3B15","rotate":"180"}' --write-out "\nhttp_code: %{http_code}\nexitcode: %{exitcode}\nerrormsg: %{errormsg}\nmethod: %{method}\n" http://10.10.10.50/save_cfg
Ok, saved
http_code: 200
exitcode: 0
errormsg: 
method: POST
```

I am using env: OpenEPaperLink_Mini_AP_v4, esp32 version: 2.60b and not able to reproduce this error.

mukowman, 2024-07-16:
Oh interesting, I thought for sure would be easy to replicate.
With the post below, including alias I get 200.
If I uncheck alias I get no response then, I check the AP and it has rebooted.

env: ESP_THREAD_BORDER_ROUTER
build date: 2024-06-06 12:43
esp32 version: custom
psram size: 2092431
flash size: 4194304

Should be 2.60b

[image]

mukowman, 2024-07-16:
[image]

LeftPinkie, 2024-07-16:
Hmmm... not sure where the issue is as I cannot reproduce. What are the headers you are sending? Is it possible to use `curl` in the terminal?

nlimper, 2024-09-04:
See https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/ESP32_AP-Flasher/src/web.cpp#L348-L382
At least parameters `mac`, `contentmode`, `alias` en `modecfgjson` must be present. Their existence is issumed, and their contents are not sanitized any further. If you like, you can add some checks, but imho, it's not really beneficial.

nlimper, 2024-10-28:
fixed in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/5e9238835e7b30e0d468a9f5fca919d5f84de0ff
`contentmode`, `alias` and `modecfgjson` are now optional.

closed 2024-10-28 by nlimper