OpenEPaperLink/OpenEPaperLink #352: Feature request:

BigSpice, 2024-07-15:
**Is your feature request related to a problem? Please describe.**
Currently have the Very popular _WEMOS D1 MINI_ and an _ESP32 - WROOM-32_ Boards.
Access to S2 based boards in my area is unavailable for various reasons.

**Describe the solution you'd like**
Possible support for the _ESP32_ or _D1 MINI_ based boards in the flashing of the Tags.
These are one of the more abundant board types, I understand implemetation would expand the project further.

**Describe alternatives you've considered**
Fully understanding of any technical limitations due to either communication or broadcast hardware.
Use of an _FTDI_ flasher to make use and edit the firmware flasher code to upload and work on a _D1 MINI_.

**Additional context**
Any help toward walls I am facing or dropped research material toward this topic would be the best and most appreciated response.

jonasniesner, 2024-07-15:
Do you want to use those boards as a Flasher or AP?

BigSpice, 2024-07-15:
> Do you want to use those boards as a Flasher or AP?

Hopefully the D1 Mini as my flasher and the ESP32-WROOM as the AP.

BigSpice, 2024-08-19:
Above was scrapped.

Unrelated but unsure where to contribute ongoing research to the community.
Going to append here.
New tag types.
More to come.
## FCC: 2AFWN-EL016H4WRC
### MODEL: EL016H4WRC
[image]
[image]
[image]
[image]
[image]
## FCC: 2AFWN-EL022F3WRA
### MODEL: ELF22F3WMA
_Unfortunately the SOC is locked down according to @LeftPinkie._

[image]
## FCC: 2AFWN-TG-GR22
### MODEL: TG-GN22F
[image]
[image]

nlimper, 2024-09-04:
New tagtypes are always welcome :-)
You can do a pull request in case of enhancements, or add them to the [wiki](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki) if you want to document stuff. 
(I'm closing the issue, as it's not an issue ;-) )

closed 2024-09-04 by nlimper