OpenEPaperLink/OpenEPaperLink #355: Solum 5.25

vaughnlfisher, 2024-08-11:
Hi again

After a lot of messing around, I uploaded the following bin from another post, m3 4.2 Firmware. https://github.com/jjwbruijn/OpenEPaperLink/files/14625655/varuicrdump-4.2-ssd-experimental.zip
It has worked on a portion of the screen, and its upsdie down. But very positive.
Could you explain why this worked an none of the other FW bins I uploaded. Attached my dump
working_infopage.zip

[image]

nlimper, 2024-09-04:
The 5.85" is not supported yet, but the fact that most of it works is very promising :-)
I will close this one, as it is a duplicate of https://github.com/OpenEPaperLink/OpenEPaperLink/issues/304

closed 2024-09-04 by nlimper

MaxwellJK, 2025-10-02:
hi @vaughnlfisher I have the same tag. May I ask you what procedure you followed to flash it?
Thanks