OpenEPaperLink/OpenEPaperLink #356: Help offer: Add support for Chroma 1.6/2.1/3.7/6.0" tags

zendes, 2024-08-11:
I've recently acquired a batch of Chroma SubGhz tags, and noticed that not all of them have firmware available. I have successfully flashed Chroma 2.9, 4.2 and 7.4" tags, connected to a homebuilt Yellow AP with an EByte CC1101 module, but also have one of each of the following sizes: 1.6", 2.1", 3.7" and 6.0". I also got a different one called Aura 42, which seems to use the same pcb but a different epaper display. I'll attach photos of all the internals.

If people (probably @skiphansen) are interested in working together on adding support for these tags to the codebase, I'm happy to help. I have debug equipment such as a logic analyzer and can pull/build code, but I haven't worked with this microcontroller or epaper displays before. Let me know if this is something you are interested in and have time for.

[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

skiphansen, 2024-08-12:
@zendes Yes, I am very interested in working with you!  I have a number of Chroma 21s and that is my next target, but you have several other tags that I don't have examples of.  Let's chat on discord.

skiphansen, 2024-08-12:
To get you started here's some [documentation](https://github.com/skiphansen/dmitrygr-einkTags/blob/master/docs/Chroma29.md#logic-analyzer-connections-for-epd-reverse-engineering) on how I connected my logic analyzer to the Chroma29 when I reverse engineered it.

I recently bought a "spider board" to make it easier to reverse engineer tags without a bunch of soldering. I'm waiting for it to arrive before tackling the Chroma21.

The first thing to do is backup the stock firmware.  This is easy since the device is not read protected.  To void possible copyright issues please do not post them publicly.  Once backed up please add the MD5 signature of the stock fw to the Wiki [here](https://github.com/OpenEPaperLink/Tag_FW_Chroma/wiki/Known-stock-firmware-images).

The next thing to do is to backup the factory EEPROM image.  If you run Linux it's easy "ish" using my chroma_shell tool and proxy firmware. Documentation is [here](https://github.com/skiphansen/dmitrygr-einkTags/tree/master/chroma_shell#readme).

zendes, 2024-08-13:
thanks. I'll try to dump the firmware on all of them this weekend, and report back if I have any issues. so far I've just opened them to see the logic board, and all of them are still running stock firmware

pandaboy6621, 2025-02-11:
Any updates for the Chroma 21? I have two of these tags.

skiphansen, 2025-02-11:
@pandaboy6621 I got distracted by other projects.  You're only the second person that I know of that has any, if more people
had them it would be more worthwhile.

uszke, 2025-10-15:
I was planning on selling them, but my morals don't allow me to do that if I'm not certain that every piece is working and I don't have the original programmer to check that easily.
That being said, I'm pretty sure that I can share some for free with someone that's interested in porting them.
We can arrange something on discord I guess?

skiphansen, 2026-10-21:
@uszke Are you still interested in porting OEPL to these tags?  

With recent advances using Ai (primarily Claude) we now understand a lot more about Chroma Tags.  New ports should be **MUCH** easier than they have been in the past.

I'm currently working on https://github.com/OpenEPaperLink/Tag_FW_Chroma/pull/11 so it would be a great time to add support for Chroma 37 and Chroma16 tags.  I started a port for Chroma21s a while back, but never finished it.

There's a good chance we could get the Chroma37 and 16 working remotely if you have time/interest.  Otherwise I'm open to donations.

skiphansen, 2026-10-22:
@zendes The last comment was really meant for you.  Are you still interested in adding OEPL support for the Chroma37 and 16?

knyghtryda, 2026-10-23:
@uszke @zendes  I ported Chroma74 to Chroma60 (see @skiphansen's message above) in a relatively short time with Claude Code.  The main things needed were the SPI dump and a firmware dump (if possible), as well as confirmations on model numbers for the display, the SPI, and the MCU.  I'm happy to do a blind port for testing if someone wants to send me some dumps.  I could also do the port directly if I had a device, but I haven't seen any of your devices for sale (yet) anywhere.  It looks like both of you are probably not US based, so sending devices might not be a great option.

zendes, 2026-10-23:
I am US based (west coast, CA). I've been AWOL bc of family reasons, but I can at least dig up the devices and send them to someone if that helps. I just moved last week, but I kinda know in which moving box they are :-)

knyghtryda, 2026-10-23:
@zendes Awesome!  I'm in socal.  Let me know if I can reach out to you via your posted email address.

skiphansen, 2026-10-24:
@knyghtryda I'm also in socal, Palos Verdes to be specific.  I have way more Chroma21s than I need, perhaps we can meet for lunch or something and I can give you a few.  @zendes if you local and have time maybe we can have an impromptu OEPL conference!  

I already have a port for the Chroma21 that I've been playing with on an off for a couple of years, but it's not quite right.  The problem is the yellow layer isn't working correctly.  I can commit that code to give you/Claude a staring point.

Please DM me on Discord if you are interested.

skiphansen, 2026-10-25:
@knyghtryda Here's an EEPROM dump of an Chroma21: [JG10014896_eeprom.zip](https://github.com/user-attachments/files/33926944/JG10014896_eeprom.zip)

decode_factory.py successfully parses NVRAM, but not the temperature table.

I believe the 2 bytes @ 0x2004 are the data format version number.
For the Chroma60 and it's 02 03 but for the Chroma21 it's 04 02

Chroma60:
```
<magic> "beafface"
00002000  ce fa ef be 
<version>
02 03 
<description: 27 bytes>
38 31  35 39 76 32 45 50 44 20  |......8159v2EPD |
00002010  50 31 35 44 30 32 2e 30  30 20 55 43 20 36 30 45  |P15D02.00 UC 60E|
00002020  34 
<unknown: 9 bytes>
00 00 00 00 65 1c 26  e4 00 
<offset base: 202a>
21 00 00 00 202a + 21 = 204B  init
```

Chroma21:
```
<magic> "beafface"
2000 ce fa ef be 
<version>
04 02
<description: 27 bytes>
44 31 38 59 30 30 30 31 30 57  ......D18Y00010W
2010 59 30 30 35 5f 55 43 38 31 35 34 20 32 31 45 34  Y005_UC8154 21E4
2020 00
<unknown: 9 bytes>
00 00 00 00 85 06 31 8a 00 
<offset base: 202a>
00 00 0d 12 00 0d  .......1........
```

Since you already have a Claude context perhaps Claude can figure out this format.

knyghtryda, 2026-10-25:
@skiphansen Here's what claude found for the chroma21 formats, as well as a decode script.
[CHROMA21_WAVEFORM_FORMAT.md](https://github.com/user-attachments/files/33938888/CHROMA21_WAVEFORM_FORMAT.md)
[decode_chroma21.py](https://github.com/user-attachments/files/33938889/decode_chroma21.py)

skiphansen, 2026-10-25:
@knyghtryda Cool and amazing!  Thanks !!!
I'll give it a try as soon as I get a chance.

skiphansen, 2026-10-26:
@knyghtryda Here's another mystery for Claude to solve.

I know that at least one [version of the firmware]( https://github.com/OpenEPaperLink/Tag_FW_Chroma/wiki/Known-stock-firmware-images) that supports devices with serial numbers of JA0xxxxxx, JA1xxxxxx and JG1xxxxxxx.  I suspect it also supports the rest of the models that existed at that time.

So the question how does the firmware know if the [Display enable](https://github.com/OpenEPaperLink/Tag_FW_Chroma/wiki/Chroma-Series-Pin-Connections) FET is present on the board or not.

It is critical for the EPD pins to be floated or driven correctly during sleep to minimize battery current.
The current firmware parses the SN to make this determination but this is probably not the best way.  In fact my first guess was wrong which caused very excessive battery current during sleep on certain SN ranges (see https://github.com/OpenEPaperLink/OpenEPaperLink/issues/459).

Additionally Claude has located the screen resolution in NVRAM (type 6) but how does the firmware know what the EPD command byte is?
Since all known controllers use  TRES (R61h) perhaps the value was just hardcoded into the firmware.

If you are in the mood to ask Claude to just write a driver that would be awesome!

skiphansen, 2026-10-29:
@knyghtryda Don't bother Claude with this one.  I figured out a way to test the HW at run time by using the pullup resistor that's present on the nRST pin of the EPD in both cases.

That code (along with your NVRAM fixes) will be in the next release so I can stop parsing the SN and guess what SN range has power control.

zendes, 2026-10-29:
I'm in the Bay Area, so a trip to Palos Verdes is about 6 hours. I'm happy to mail the units to either of you, so just let me know who to send them to and a way to get your address.

skiphansen, 2026-10-30:
@zendes @knyghtryda Look at each other's github profiles and I bet you'll find a way to communication privately 😉