OpenEPaperLink/OpenEPaperLink #360: S2 Tag Flasher: Installation failed ⚠️ Timed out waiting for packet header

bonsai308, 2024-08-15:
When trying to flash an S2mini via install.openepaperlink.de with the "Tag Flasher" the following error message appears 99% of the way through the process:

Installation failed
⚠️
Timed out waiting for packet header

The DFU mode on the S2mini works. For example, I was able to transfer ESP-Easy to the S2 mini via another ESP Flashing website. For testing purposes I also tried holding down the boot (O) key. But this makes no difference.

Are the following files available somewhere:
0xe000 -> boot_app0.bin
0x1000 -> bootloader.bin
0x10000 -> S2_Tag_Flasher_full.bin (this is available in the repo)
0x8000 -> xxx.ino.partitions.bin
Then I could try it manually.

jonasniesner, 2024-08-15:
If it failed at 99% it most likely worked and you just need to reboot the esp

bonsai308, 2024-08-15:
In my case, unfortunately, nothing happens after the reset.
I followed the instructions and triggered a reset, then reloaded the website and pressed the tag flasher again. But then no connection is established.

One question by the way: does the firmware open an access point to configure the WLAN or can this only be done via the console or web flasher?

jonasniesner, 2024-08-15:
The S2 tag flasher does not open an AP, nor can it be used with the install page, it has to be used over USB with the flasher python script

closed 2024-09-29 by jonasniesner

linesquarecube, 2025-04-04:
Hi there,
could this possibly be highlighted somewhere? I'm not an expert, and I was quite confused. Luckily, I stumbled upon this issue. Connection to flasher works perfectly even with the error, just as you said 😃