OpenEPaperLink/OpenEPaperLink #364: Feature request: Eink 23"

SvetZitrka, 2024-08-29:
I would like to use Home Assistnat to share tasks in the family through a large Eink board.

The biggest one we found is this one https://www.ebay.com/itm/286013702072

Would OpenEPaperLink be able to control it?

atc1441, 2024-08-29:
the linked display is a segment display only. 

No images can be displayed on it. 

Because of that it makes no sense to support it

closed 2024-09-04 by nlimper

SvetZitrka, 2024-09-04:
What is the largest eink display available so far where images can be displayed?

nlimper, 2024-09-04:
Which is supported by OEPL? That might be the Solum M3 12.2" 
But that's all in theory, because the bigger displays (or, even the smaller displays) are almost impossible to find.

atc1441, 2024-09-04:
@SvetZitrka feel free to send me a Mail to atc1441@gmail.com for a 12,2" variant, got few of them