OpenEPaperLink/OpenEPaperLink #367: Acces point says it's offline, then it says it's failed

Rajmi105, 2024-09-02:
So when I flashed my ESP32-S2 with the Mini-AP v2, the installation went fine, I was able to connect it to my wifi, but then I see that it says that it is offline, than after some seconds it will say that it's failed.

I also looked at the terminal, and it said ping failed

I reflashed my ESP32-S2 multiple times, and I don't know how it works, please help

nlimper, 2024-09-04:
If it shows the 'failed' status, the esp32 cannot communicate with the radio module. I assume you connected a segmented tag or an esp32-C6, but in any case, check your wiring first. A segmented tag should be flashed automatically (at least, you will see attempts in the serial output), a esp32-c6, you have to flash yourself (using the 'update c6' button in the update screen).
It's not a bug, you just didn't connect the right stuff ;-)

closed 2024-09-04 by nlimper

Rajmi105, 2024-09-15:
I didn't connect a segmented tag to my ESP32 because I didn't think it was needed to make the access point usable, I just flashed my board with the software and tried to use it, because on paper the ESP32-S2 should have everything in itself that is needed for this to work.

Is it neccessary to use an e paper tag with it? If so than sorry, for not using one

nlimper, 2024-09-15:
An esp32-s2 cannot talk IEEE 802.15.4, so that's why you need two MCU's.
Also, esp32-s2 has barely enough memory, so it's advisable to use a esp32-s3 (16MB flash, 4MB psram) plus a esp32-c6.
You can find all info on how to build an access point the wiki: https://github.com/OpenEPaperLink/OpenEPaperLink/wiki