OpenEPaperLink/OpenEPaperLink #371: Feature request: Emulator for an ePaper display (virtual display) ?

gitpower2017, 2024-09-24:
Especially the development of a nice display with json commands requires many updates of a display. 
If you want to write a visualisation for a display that you do not own yourself, this is also difficult or even impossible

Not only does such an update take a relatively long time, it also drains the battery life.

There is already a small preview in the Webgui, perhaps it can be extended to a virtual display?

jonasniesner, 2024-09-24:
I was thinking about this feature for the home assistant integration. Do you use Home Assistant or just the AP directly?

gitpower2017, 2024-09-24:
I use the AP directly.
A solution in the AP is also much more flexible, you can manage with fewer dependencies. Everyone benefits from it.

closed 2024-10-28 by nlimper