OpenEPaperLink/OpenEPaperLink #372: Feature request: websocket sys message additional infos

micklafisch, 2024-09-25:
**Is your feature request related to a problem? Please describe.**
the sys message on websocket connection tells unfortunately not if are there offline/pending or low-bat devices only the total count

**Describe the solution you'd like**
maybe add the counter for offline/pending and lowbat devices, with these informations iobroker can send i.e. an alarm with telegram adapter

nlimper, 2024-10-28:
in commit https://github.com/OpenEPaperLink/OpenEPaperLink/commit/0cd76eebf0214665276962b220070e7e6643161f

I added the number of timed out tags to the 'sys' websocket message. The timeoutcount key/value is only present once a minute.
`{"sys":{"currtime":1730133055,"heap":194124,"recordcount":27,"dbsize":5488,"littlefsfree":5246976,"psfree":8350371,"apstate":1,"runstate":2,"rssi":-70,"wifistatus":3,"wifissid":"sensors","uptime":72,"timeoutcount":0}}`

For the pending/lowbatt: you can find those already in the individual tag messages in the websocket stream. Note that having a pending image is a normal state, so no need for a notification.

closed 2024-10-28 by nlimper

micklafisch, 2024-10-29:
Thanks Nic for implementation.

Perhaps it is possible after all to add the lowbatt information to "sys", Otherwise i have to watch all incoming tag messages and with more than 20 tags there is lot of data only for the lowbatt counter.

nlimper, 2024-11-02:
added the lowbatt count in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/a57c797542536004cba3298d91e1668260b49106
It will report back the amount of lowbatt tags at the same time as the timeoutcount, so once a minute.