OpenEPaperLink/OpenEPaperLink #376: Gicisky 2.13" TFT display garbled/distorted

MrDeon, 2024-10-02:
**Describe the bug**
I'm attempting to send a basic image from OpenEPaperLink on my ESP32-S3 to a Gicisky 2.13" TFT shelf label. No matter what I try (static image, display date, AP info, etc), it is always garbled and unreadable. I have tried flashing the firmware again on the tag (using TLSR_Gicisky_TFT_213.bin) and also updated the AP to 2.60b, but this didn't help. I'm using BLE_ONLY_AP as the Gicisky tags are BLE. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Open EPaper Link Access Point web interface
2. Select the tag
3. Set Content to 'Static image', draw a basic image and click Save
4. Wait for the tag to get the update
5. Observe that the image on the tag looks nothing like the image drawn in the web interface

**Expected behavior**
The image on the tag to match the image drawn in the web interface (or to match whatever other content like current date, QR code, etc)

**Screenshots**

[image]
[image]

MrDeon, 2024-10-02:
Oh, I didn't realise it was unsupported. I picked the first one under **BLE based** on [this page](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki#tags) which is the Gicisky 2.13" TFT.

Is there another compatible BLE tag you can recommend I replace this one with?

closed 2024-10-05 by atc1441

MrDeon, 2024-10-12:
Hi @atc1441 - sorry to bring you back to this topic again

I couldn't find any of Gicisky Epaper tags available in my country, so I have bought an ESP32-C6 and flashed both ESP32s in the 'spaghetti AP' configuration

Can you suggest where I can find the Zigbee firmware for the Gicisky 2.13" TFT tag? I wasn't aware this tag had a Zigbee module/radio in it, but if this is indeed the case, I'm hoping I can get it working with the setup I have now.

atc1441, 2024-10-13:
You can use this tool to flash 
https://atc1441.github.io/BLE_EPaper_OTA.html
And the firmware is here
https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/binaries/Tag/TLSR_Gicisky_TFT_213.bin