OpenEPaperLink/OpenEPaperLink #380: Day ahead electricity price shows "nan" instad of current price per kWh

mountaindude, 2024-10-06:
**Describe the bug**
Instead of showing current cost/kWh, the display shows "nan/kWh"

[image]

**To Reproduce**
[image]

C6 chip updated to latest version.
Tag type definitions updated.

**Expected behavior**
Up until a week or two ago the correct, current price was shown in the display.

**Additional context**
I would guess the issue occurs [here](https://github.com/OpenEPaperLink/OpenEPaperLink/blob/8986967a4d364618c979a3636ecea44a21dd8df5/ESP32_AP-Flasher/src/contentmanager.cpp#L1575) somewhere.

nlimper, 2024-10-06:
Thanks for your report. The upstream API provider (ENTSOE) has unexpectedly migrated their transparancy platform to a new server (https://transparency.entsoe.eu/news/widget?id=66e3ecae61d4cd1be808f94d). I haven't found the new API entry point, will dive into it in the next few days. When fixed, your display will start working again without intervention. Please stand by. :-)

AseKarlsson, 2024-10-06:
Thanks @nlimper!

nlimper, 2024-10-28:
After a few days, the error solved itself, probably after ENTSOE debugged their migration.
Can you confirm it's also working again for you? (if not, please tell me which country)

mountaindude, 2024-10-28:
Yup, works here in Sweden. 
That was a convenient solution, things fixing themselves :)

closed 2024-10-28 by mountaindude