OpenEPaperLink/OpenEPaperLink #383: Gicisky 2.13 inch BLE Tag Incorrect Size

Jestinator, 2024-10-25:
I have a Gicisky 2.13 inch TFT BLE Tag which is found by the AP fine but the images are distorted.

I've found that the image being sent is 250x132 however the display is 250x122.

I've tried editing the tagtype file BA.json on the AP and it shows 250x122 on the tag info screen but I can see in the log on the device that the image is encoded to 250x132 regardless.  This results in an unreadable display.

atc1441, 2024-10-25:
Hey, this display is not fully supported right now.
This is the expected outcome

jonasniesner, 2024-10-26:
I updated the wiki to avoid confusion in the future

closed 2024-10-26 by jonasniesner

atc1441, 2024-10-26:
Thank you @jonasniesner

azdlowry, 2024-12-10:
Is there a specific issue making the 2.13 inch difficult to support or is it just that no one has looked at it yet? I bought the wrong ones accidently and have some time to look at fixing them at the moment.

atc1441, 2024-12-10:
Hey @azdlowry  you can flash this OEPL Firmware onto it https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/binaries/Tag/TLSR_Gicisky_TFT_213.bin
via this flasher: https://atc1441.github.io/BLE_EPaper_OTA.html
It will then work on the OEPL 802.15.2 Protocol and is working fine.

azdlowry, 2024-12-11:
Thanks @atc1441 .
I'm running a BLE only AP at the moment. I'll have a poke around while I'm looking at some other issues.

connoreaton3, 2025-02-11:
I am currently running BLE only AP on a esp32-s3 and have flashed the firmware on my tag but now it just shows No AP Found. Which AP should I be using for this?

Obamium69, 2025-08-03:
Hi @connoreaton3.

I also having the same setup and the same issue. Did found a fix for that?

connoreaton3, 2025-08-10:
> Hi [@connoreaton3](https://github.com/connoreaton3).
> 
> I also having the same setup and the same issue. Did found a fix for that?

Hi @Obamium69,

Yes, I managed to get it all working. It was a bit of a pain and had a few issues whilst doing so, but please see below where I posted my full fix and steps taken to get it all working.

https://github.com/OpenEPaperLink/OpenEPaperLink/issues/437#issuecomment-2866210027