OpenEPaperLink/OpenEPaperLink #385: ZKong/Goodix GR51xx based tags

silverchris, 2024-10-26:
Hey Everyone,
Didn't see a more appropriate place to ask this. I have come into a number of these from a business closure, and am trying to repurpose these. 
Before I get too deep into reverse engineering them myself, has anyone already started this process?

They are to be Bluetooth based, at least the ones I have looked at so far.

atc1441, 2024-10-26:
Hey. These are not supported or looked into so far

silverchris, 2024-10-26:
Cool, if I find out anything useful, I will try and contribute it or at least notify about it! 
I am hoping to figure out the protocol so they don't need to be reflashed, but that is being optimistic. Hoping I can dig out the configuration of the display and such as well.

atc1441, 2024-10-26:
Sounds good 

Only thing to note here. We in this project only talk and will do anything custom and nothing stock firmware or Protocol related so a reflash is always needed.
We do not want to annoy any manufacturer with our tinkering.

silverchris, 2024-10-26:
Of course! I will keep the reverse engineering in it's own repo, and just pass back the useful hardware stuff for OpenEPaperLink.

Hoping there is a non-destructive way to reflash them, as the larger display I opened has the test points facing he screen, and it did not disassembly easily. 

The Goodix chips do seem to have a bluetooth firmware recovery method in the rom, though, so I have some hope for that.

closed 2024-12-17 by atc1441