OpenEPaperLink/OpenEPaperLink #386: Possible problem with "Nightly reboot"

jurasjo, 2024-10-29:
**Describe the bug**
It is a continuation of this issue: [https://github.com/OpenEPaperLink/Home_Assistant_Integration/issues/188](url)

I think, because of "Nightly reboot", jobs/automatons might not be properly sent to tags if those are scheduled at this time. 
Now it is clear why I have been observing problems around that particular time.

Looks like this reboot functionality might have something to do with that. Maybe it should be optional or configurable, so it can be set at a time, when there is no planned job.

Full hour might not be the best time for reboot.

**To Reproduce**
Steps to reproduce the behavior:

Schedule jobs/automations in HA which run around 4:00 and there is a big chance it will not be properly sent to AP/tags, because AP is not available for about 30 sec to 1min at that time.

**Expected behavior**

Reboot of the AP should not interfere with sending data to tags.

nlimper, 2024-10-29:
I will make it optional, and use 3:56 instead of 4:00

jurasjo, 2024-10-29:
This sounds great. Thank you.

nlimper, 2024-11-02:
done, in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/a57c797542536004cba3298d91e1668260b49106

closed 2024-11-02 by nlimper

jurasjo, 2024-11-03:
Thanks for that. So in order to apply this change, the new release must be available?

nlimper, 2024-11-03:
That's right. There will be a new release shortly.