OpenEPaperLink/OpenEPaperLink #389: [Cosmetic] Update page shows version discrepancy

Calimerorulez, 2024-11-05:
Just a cosmetic bug I think. I just updated to 2.70, and now the update page shows:

```
Your version is newer than the latest release date.
Are you the developer? :-)
```

```
env:                ESP32_S3_16_8_YELLOW_AP
build date:         2024-11-03 20:31
esp32 version:      2.7
filesystem version: 2.70 <- not matching!
psram size:         8383831
flash size:         16777216
--------------------------
```

Same on a v3 AP.

Calimerorulez, 2024-11-05:
Addition:

I've got 4 APs, 2 are updated to 2.7, but the Active access point page shows all APs are at 2.7.

nlimper, 2024-11-05:
your first point: just cosmetic, the 2.70 is treated as numeric somewhere and displayed as 2.7
Already fixed in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/7805ab2b467f71bb94e1aed650b48e7847f62294

But I leave this open for your addition. I will look in to it.

nlimper, 2024-11-08:
Fixed in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/a6772c6fae3a4561c6f350246bf9f52600a52345
It was a frontend problem. To fix, replace `/www/main.js.gz` with the updated file (and make sure it's refreshed from the cache in your browser).