OpenEPaperLink/OpenEPaperLink #395: Feature request: Digi

chrisdal2000, 2024-11-11:
Can any help me with that?
https://device.report/m/6c35165a563ee24126330f61bff159a5cdf49a7641bd2119886c686324ca0b5f.pdf

Is a https://device.report/digi/ product.
How i can put a new firmare and then can upload a image from a esp32?

thanks

jonasniesner, 2024-11-11:
Those look nice because they have BWRY. How many of them do you have?

chrisdal2000, 2024-11-11:
hi, many (20+).
They have a program for sending the logo but not like and i want to make i new if i can,

jonasniesner, 2024-11-11:
https://github.com/OpenEPaperLink/Tag_FW_EFR32xG22 would be a good starting point. You can also try to ask around on the discord server.

chrisdal2000, 2024-11-11:
Thanks for your time.

Can you have a link for discord server. I dont use it.

jonasniesner, 2024-11-11:
https://discord.gg/eRUHt4u5CZ

chrisdal2000, 2024-11-12:
Do you know where to buy new epaper same that (from ebay, amazon ...) and can put i firmware and can then upload the image with vb.net + esp32 ?
thanks

closed 2024-12-17 by atc1441