OpenEPaperLink/OpenEPaperLink #403: missing esp_ieee802154_receive_handle_done() function call causing stop of ZigBee reception

m-meltner, 2024-11-25:
Describe the bug
After 20 receptions on OpenEPaperLink_esp32_C6_AP no more packets are received over ZigBee. All previous packets arrive just fine.

To Reproduce
Steps to reproduce the behavior:

- reboot OpenEPaperLink_esp32_C6_AP
- reset tag so it registers
- count received ZigBee packets
- after the 20th packet no more packets arrive

Expected behavior
continuous reception of packets

Additional context
See: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/migration-guides/release-5.x/5.1/ieee802154.html

Fix:
I added a call to esp_ieee802154_receive_handle_done() in function esp_ieee802154_receive_done in file "radio.c". This makes it work just fine.

- Michael
-

skiphansen, 2024-11-25:
I think this is related to the version of IDF used.  The auto builds use "the latest" and do not have this issue.  What version of IDF do you use?

skiphansen, 2024-11-26:
The binaries are here: https://github.com/OpenEPaperLink/OpenEPaperLink/tree/master/binaries/ESP32-C6

You can use the esptool to flash.  I find that copying the command line that IDF uses when flashing and then modifying the arguments is easiest for me.  In my case it looks like this:

```
esptool.py --chip esp32c6 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x0 bootloader/bootloader.bin 0x10000 OpenEPaperLink_esp32_C6.bin 0x8000 partition_table/partition-table.bin

```

Thanks for the kind words about the project, but I'm not a primary author. Jelmer, nlimper, atc1441, and jonasniesner are the pillars of the project.

I'm a bit player who added SubGhz support, and probably the one that screwed up the C6 code.

skiphansen, 2024-12-07:
@m-meltner I've verified the problem and fix locally.  It turns out that this bug was introduced in version 0x001d but the version installed by the WEB GUI is 0x001c which explains why most people are not experiencing this issue.  I'll commit a fix shortly.

Thanks for reporting the issue and fix !!

commit 1871f53b (2024-12-07): Fix #403: missing esp_ieee802154_receive_handle_done() function call causing stop of ZigBee reception.
closed 2024-12-07 by skiphansen