OpenEPaperLink/OpenEPaperLink #405: Solum 5.85-inch Display Not Fully Utilizing Screen with Google Calendar Script

MagnusErler, 2024-12-07:
**Describe the bug**
The Solum 5.85-inch e-paper display does not fully utilize the screen area when running the [Google Calendar script](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Google-Apps-Scripts#google-calendar). Portions of the display remain blank instead of being filled with content.

**To Reproduce**
Steps to reproduce the behaviour:
1. Flash the tag with the firmware: [SOL_M3_Uni_ota_0029.bin](https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/binaries/Tag/SOL_M3_Uni_ota_0029.bin)
2. Set up and deploy the [Google Calendar script](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Google-Apps-Scripts#google-calendar) as per the instructions
3. Upload the script to the tag
4. Observe the rendered display on the 5.85-inch screen

**Expected behaviour**
The entire screen area of the Solum 5.85-inch display should be filled with the content provided by the Google Calendar script.

**Actual Behavior**
The content does not fully occupy the display area, leaving portions of the screen blank.

**Picture**
[image]

nlimper, 2024-12-08:
The 5.85" currently doesn't have its own templates for the resolution, so the content is not optimal. I sadly don't own a 5.85", so I cannot work on it. To create the templates, you can edit the /tagtypes/41.json file, and add a `"template": {}` section with the right coordinates for the content types. It can be a bit fiddly to do, but you will get a nicer layout. (after you edit a tagtype json file, you may have to restart the AP for it to work).
Instead, you can also alter in the 41.json file: `"usetemplate": 47`, by doing that, it uses the 4.3" tag as a fallback, which is more fitting for a start.

MagnusErler, 2024-12-08:
Thanks for the guidance. I will try editing 41.json

nlimper, 2024-12-17:
fixed in https://github.com/OpenEPaperLink/OpenEPaperLink/pull/406
Thanks for fixing!

closed 2024-12-17 by nlimper

MaxwellJK, 2025-10-02:
Hi @MagnusErler I have the same tag - may I ask you what procedure you followed to flash it?
Thanks