OpenEPaperLink/OpenEPaperLink #410: Tags not updating after ESP32_C6 re-flash with latest binary

Rudios81, 2024-12-11:
**Describe the bug**
After updating my AP to my own recompiled environment, I had issues with updating the C6 chip via OTA.
I finally managed to download the binary file from the master repository, and uploaded it onto my AP. From there I managed to update the C6 chip.
Unfortunately, none of my tags updated anymore after this update!
I have been trying to update the C6 chip many times, always got stuck at the download actions.

Finally I downloaded an older version of hte ESP32_C6 binary, the one from release 2.60 and uploaded this into my AP.
After updating the C6 chip with this binary file, some of my tags started to update immediately.

collateral87, 2024-12-12:
I can confirm this issue. I flashed my C6 with the binary from version 2.7 and instantly the tags are working again.

skiphansen, 2024-12-14:
Thanks for the report!  I've posted a new C6 image to the help_and_questions topic on Discord, please give it a try and let us know if it resolves the problem or not.

Thanks!

Boorte, 2024-12-15:
I can also confirm the issue, however with some additional issues. I updated to latest FW on my yellow AP and all tags stopped working (was working fine before), however rolling back to previous FW did not make the tags work again. I've been trying to upgrade/downgrade, update filesystem and tag definitions/DB but nothing seems to help. The status of the AP says failed.

nlimper, 2024-12-15:
If the status of the AP shows 'failed', don't mess with the ESP32-S3 firmware, filesystem, tag, or other things. In that case, focus on the ESP32-C6 firmware, because the 'failed' status indicates that's not working correctly.

Boorte, 2024-12-15:
The ESP32-C6 update failes with the message "Packet could not be written!" and "Flashing failed. :-(".
I guess I have to find the old C6 firmware from 2.60b as well

Rudios81, 2024-12-16:
> Thanks for the report! I've posted a new C6 image to the help_and_questions topic on Discord, please give it a try and let us know if it resolves the problem or not.
> 
> Thanks!

I can confirm that the version supplied on the discord discussion does solve the problem!

skiphansen, 2024-12-16:
Thanks for reporting the problem and testing the fix!

commit bb361850 (2024-12-17): C6 Ver 001f - fix #410 & other C6 improvements

1. Only call esp_ieee802154_receive_handle_done() when ack != NULL.
2. Add 2 millisecond timeout for all SubGhz MISO wait loops to
   prevent watchdog timeouts when/if bad things happen (none observed
   during testing)
3. Make CC1101 detection more robust and less intrusive by testing
   MISO and CSn before trying to read chip version number.
4. Remove useless "error" messages which occur during normal operation.
5. Added missing \r in some log messages for EOL consistency.
closed 2024-12-16 by skiphansen