OpenEPaperLink/OpenEPaperLink #422: Unable to fetch ESP32-C6 firmware because of different filename

rautmithil, 2024-12-28:
firmware.json in that folder is named firmware_C6.json so update process of C6 via web gui fails.
Log

`Flashing ESP32-C6...
Stopping AP service
C6 flash starting
http error 404 fetching https://raw.githubusercontent.com/OpenEPaperLink/OpenEPaperLink/master/binaries/ESP32-C6/firmware.json
C6 flash end
Flashing failed. :-(`

rautmithil, 2024-12-29:
Solution is to update these two lines with filename firmware_C6.json

https://github.com/OpenEPaperLink/OpenEPaperLink/blob/ca8781f95622a97eab57458ee110b81e2ddb6ec2/ESP32_AP-Flasher/src/espflasher.cpp#L171

https://github.com/OpenEPaperLink/OpenEPaperLink/blame/ca8781f95622a97eab57458ee110b81e2ddb6ec2/ESP32_AP-Flasher/src/espflasher.cpp#L174C121-L174C121

atc1441, 2024-12-29:
Thanks, a new C6/H2 update mechanism is being developed right now with these filename changes already made added.
Did add a copy of the old named json files for now to bypass any error

closed 2024-12-29 by atc1441