OpenEPaperLink/OpenEPaperLink #430: Chroma 2.9" tag MAC address reset to 00000000 after flashing firmware 0x10

classabbyamp, 2025-01-15:
**Describe the bug**

The tag MAC is zeroed after flashing the OEPL firmware

**To Reproduce**
Steps to reproduce the behavior:
1. Download https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/binaries/Tag/chroma29_full_0010.bin
2. Flash with a cc debugger and `cc-tool -e -w chroma29_full_0010.bin -v`
3. Reassemble tag and watch it boot
4. Notice tag MAC has changed from the number on the case to 00000000

**Expected behavior**

The tag MAC should stay consistent

**Screenshots**

[image]

I now have at least 2 tags that have the same MAC, so the AP shows them as 1 but updates them both with the same image.

skiphansen, 2025-01-18:
If you encountered this when flashing the tag for the first time the EEPROM was probably already corrupt.  We've seen this on a few TAGs recently.

I'll eventually add a Wiki page on recovering from a corrupted EEPROM, in the mean time this [Python script](https://github.com/OpenEPaperLink/Tag_FW_Chroma/tree/master/Chroma_Tag_FW/patch_sn) can be used to patch chroma29_full_0010.bin with the correct SN for your tag and then you can reflash it to recover.

Please give it a try and let me know if you have any questions.

skiphansen, 2025-01-23:
Great, thanks for the feedback. I would be interested to know what percentage of your tags had corrupted EEPROMs.

classabbyamp, 2025-01-23:
I had ~5 do it so I started using the patching script for all of them. I've flashed ~20, had one get stuck with the wrong mac (accidentally flashed with the wrong number) but otherwise successful.

skiphansen, 2025-01-24:
As you have discovered the firmware only resets the SN if it's corrupt or the default, once it has a valid SN an update with a different SN is ignored.

We can recover that one as well, but not easily.  My [chroma_shell](https://github.com/skiphansen/dmitrygr-einkTags/tree/master/chroma_shell) utility could be used to erase the first sector of the EEPROM and then you could reflash with a patched image again and you should be good to go.

If you want to do that you'll need a Linux system and a serial port connected the tag.

skiphansen, 2025-02-01:
I've added documentation to the [Wiki](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Restoring-Lost-Serial-Numbers-in-Chroma-Tags).  Closing issue.

closed 2025-02-01 by skiphansen