OpenEPaperLink/OpenEPaperLink #432: Dayahead prices not working

dr3as, 2025-01-20:
From yesterday my tag for showing the prices for electric is not working. The tag shows nan/kWh for the prices.

**Additional context**
I'm in norway, and uses the prices for NO2

dr3as, 2025-01-20:
This was most likely something wrong on the site hosting the information, as this now is fixed.

closed 2025-01-20 by dr3as

nlimper, 2025-01-20:
ENTSOE (the source of the info) had a disruption. That happens sometimes, and it resolves itself.
"Due to technical issues Energy Prices 12.1.D are not available for 20th January in most of the areas. 
As soon as the issues are resolved Energy Prices for 20th January will be published on TP."
https://newtransparency.entsoe.eu/news?id=678e3a6bc3645d7db0416fbd