OpenEPaperLink/OpenEPaperLink #433: Tag display wrong - 2.13" GiSky tag

bogd, 2025-01-25:
**Describe the bug**
BLE tag seems to be (correctly?) identified, but display is wrong (rotated/multiplied?)

**To Reproduce**
I have a BLE tag of... unknown manufacturer - purchased from AliExpress. Black/white, 2.13". The tag is identified by the AP as:

Gicisky BLE TFT 2.13"
250x136 fw:33025 0x8101
RSSI -62, LQI 194, 2.9V

However, the display seems completely wrong - see screenshots below.

**Expected behavior**
Correct display

**Screenshots**
See [here](https://imgur.com/a/TipBAOC). I tried displaying a days counter (showing 0), and the current date. You can see the image as sent by the AP (screenshot taken from the AP web interface), and a photo of what the tag is actually displaying.

I am aware that the answer may simply be "get a different tag" :) , but... I thought I should post this, maybe someone has seen this problem before and has a solution.

atc1441, 2025-01-25:
https://github.com/OpenEPaperLink/OpenEPaperLink/issues/383

closed 2025-01-25 by atc1441

bogd, 2025-01-25:
My apologies, I did search before creating a new issue - apparently, not with the right keywords :)

Thank you for pointing me in the right direction!

atc1441, 2025-01-25:
No worries :)