OpenEPaperLink/OpenEPaperLink #434: websocket sys message deviating value to webinterface

micklafisch, 2025-01-29:
**Describe the bug**
i check the syslog messages for monitoring the system and a lot of "timeout" tags are reported. In the webinterface there is only 1 tag with timeout, see down below. I think the problem is related to the maximul sleep time (i set it to 30mins)

**To Reproduce**
Steps to reproduce the behavior:
1. set maximum sleep timer of AP to 30mins
2. wait a little, maybe 30 minutes =)
3. check syslog message and compare the value with the webinterace

**Expected behavior**
same values no matter where the value is queried

**Screenshots**

`{"sys":{"currtime":1738159020,"heap":178232,"recordcount":37,"dbsize":7157,"littlefsfree":5222400,"psfree":8179991,"apstate":1,"runstate":2,"rssi":-34,"wifistatus":3,"wifissid":"GHFN","uptime":39653,"lowbattcount":2,"timeoutcount":24}}
`

[image]

nlimper, 2025-03-24:
Thanks for noting.
I found the reason of the discrepancy, that's fixed (in a future commit). But some tags are marked as timed out, because they are sleeping for the 'max sleep time' (that's by design) but the expected checkin time is not correctly saved. I'm investigating.

nlimper, 2026-10-14:
This should be solved in the latest release: the expected check-in time is now based on what the radio actually told the tag (commit af343b8c), so tags that sleep the maximum sleep time are no longer expected too early. That removes most of the difference between the sys message and the web interface; for a tag on maximum sleep they can still disagree for a while (up to about 25 minutes) before both count it as timed out. Please reopen if you still see a real difference.

closed 2026-10-14 by nlimper