OpenEPaperLink/OpenEPaperLink #436: Decrease unnecessary refreshes of the tag?

classabbyamp, 2025-01-31:
I'm using Chroma29 tags. Is it possible to get the firmware to only refresh when there's actually new data, or if a certain amount of time has passed? Even when using the "Image url" mode with a high interval, some tags will refresh fairly often (every few minutes)

MaZZly, 2025-02-24:
I am adding my findings here also: My Chroma 29 tags also perform redrawing of the screen every few minutes, although there is no pending/next update.

One of the tags I have a Home Assistant automation to update at the start of every hour; the tag details in HA shows `next update: In 46 years` but still does a redraw of the same content every ~5ish minutes. I have checked traces for the automation that it only triggers at the start of the hour.

Another tag set to show the "Dayahead prices" shows the Next update timestamp to correctly happen every hour, but it also does refreshes every ~5ish minutes. Content looks identical to me 🤷‍♂ 

A third tag that I have home assistant to update every 15 minutes also does redraws in between the 15 minutes when it should perform a redraw..

The check-ins for the tags seem to happen every minute as they should. Dunno why they go ahead and redraw when no update is pending..

MaZZly, 2025-02-24:
When these redraws happen, there is nothing in the AP logs about downloading new content etc... 

So I'm guessing it is the tag's FW that does something without being in contact with the AP...

skiphansen, 2025-02-25:
It shouldn't.  The way the OEPL protocol works the AP tells the tag when to check in next it doesn't decide on it's own.

If you connect a serial port to a tag and get a log it will tell us more. 

See: [Debug Serial port Connections](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Chroma-Series-SubGhz-Tags#debug-serial-port-connections)

MaZZly, 2025-02-26:
It seems that the batteries were the problem for me. I was using some old CR2450's I had lying around while waiting for new batteries to arrive in the mail.

After swapping to fresh batteries the display doesn't do any extra redraws and only updates the screen as instructed by the AP.

I tested 2 of the "bad" CR2450s:
- Chroma 29 tag reported them as `VBat: 2608 mV` & `VBat: 2615 mV`
- With a battery tester I have they showed as `2.71V` & `2.75V`

Interestingly, the displays still worked and updated their content with these batteries, but they did extra "redraws" of the same content every few minutes. I would imagine this would accelerate the battery drain when the cells' voltage degrades over time.

@classabbyamp what are your batteries reporting? And does swapping to new cells help?

@skiphansen I have a USB-TTL module that can be jumped to do 3.3V mode, so I guess I should be able to try and check the serial output what happens with the bad batteries, only need to figure out how I can easily "attach" to TP7 while also using the old coin cells... Maybe it is easiest to sacrifice one of my displays and solder onto TP7 for debugging.

skiphansen, 2025-02-26:
The voltage reported by the Tag is measured while the transmitter is transmitting which puts about a [60ma](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Chroma-74#battery-current) load on the battery which is probably why it reports lower than the battery tester.  Unlike some other Tags the Chroma series actually has an A/D for battery voltage measurement and if the SPI flash contents haven't been lost it's even calibrate per tag from the factory.

When the battery is too low the tag will reset while trying to update the screen, but unless you have disabled it you should see the "flash" screen between the screens with user content.

MaZZly, 2025-02-27:
> When the battery is too low the tag will reset while trying to update the screen, but unless you have disabled it you should see the "flash" screen between the screens with user content.

When these redraws happen, it simply starts flashing black/white (like it always does before drawing something), and then it redraws the same content that was there before. I haven't noticed that it would show anything other in between..

closed 2025-03-19 by nlimper