OpenEPaperLink/OpenEPaperLink #437: Yellow AP & Gicisky BLE TFT 2.13" Blank Screen

connoreaton3, 2025-02-12:
I have built the yellow AP following the guide and it is working fine.

I have 3 Gicisky BLE TFT 2.13" flashed with the bin file and it shows No AP Found.

The tags appear in the AP and when I send data (e.g. current date) to them the screens go blank and nothing happens. 

Not sure what is going on and I am pretty new to this so any help would be massively appreciated.

gveloce, 2025-05-09:
Guys, I thought that Gicisky BLE TFT 2.13" epaper tags are natively supported by OpenEpaperLink without flashing. 
Can someone confirm this as I intend to buy a few for my HA projects. TIA

atc1441, 2025-05-09:
Hey @gveloce the TFT Variant is not supported without flashing

gveloce, 2025-05-09:
> Hey [@gveloce](https://github.com/gveloce) the TFT Variant is not supported without flashing

thanks @atc1441  for the light speed response! 
Good to know. 

What flashing is then required for it ? 
I saw there is a BIN file in tag binaries TLSR_Gicisky_TFT_213.bin.

Is this the one required to flash it in order to be supported?

TIA

connoreaton3, 2025-05-09:
> thanks [@atc1441](https://github.com/atc1441) for the light speed response! Good to know.
>
> What flashing is then required for it ? I saw there is a BIN file in tag binaries [...]

I managed to finally get this fixed. 

You are correct with the .bin file (https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/binaries/Tag/TLSR_Gicisky_TFT_213.bin)

I used the ATC_TLSR BLE Flasher by ATC1441 (https://atc1441.github.io/ATC_TLSR_Paper_OTA_writing.html) at first and just used BLE to connect to the display and flash the .bin file to it.

Now, at first, as mentioned above, I did end up getting things connected, but the display just displayed "No AP Found," and as soon as I sent data to it, I accidentally soft bricked it and had to buy a UART-TTL USB Adapter and use the USB-COM Flasher (https://atc1441.github.io/ATC_TLSR_Paper_UART_Flasher.html) and remove the back from the display to connect to the pins on the baord.

After I flashed it with the USB-COM Flasher, it worked perfectly, and thankfully, I have had no issues since.

gveloce, 2025-05-09:
hi @connoreaton3 

thanks for the detailed response. Glad you succeeded flashing it.
But looks quite an adventure to flash it risking rendering it a (e)paperweight (joke :D) ...... 

Thinking maybe is better avoid buying such tag model (unless there are any easier ways to make it work with OpenEpaperLink)
Any other Tag models  verified to work w/o flashing?

Cheers,
GV,

Misiu, 2025-07-08:
@gveloce I have GICIsky 2.9 inch display - https://s.click.aliexpress.com/e/_onWdB1u
Works great without flashing.

P.S. I'm not sure whether I should or shouldn't change the firmware, as I'm unsure of its benefits. I'll probably order another display and test the firmware on it.

dennyreiter, 2025-08-12:
> @dennyreiter <https://github.com/dennyreiter> can't you use a web-based
> flasher? It should just work out of the box. All you need is Bluetooth on
> your pc/phone.
> I found flashing by phone more reliable.

I did do the web flashing. And the first time I try and send something from
the AP to the tag, it bricks and the screen shuts off. This happened on two
different tags, just like  @connoreaton3
<https://github.com/connoreaton3> described
in the issue.