OpenEPaperLink/OpenEPaperLink #442: Chaos in hours in the Dayahead prices

dr3as, 2025-02-22:
**Describe the bug**

When using the dayahead prices on my tags, i get a strange error. In the bottom of the graf is says the hours, and it starts right with f.eks 12-14-16 etc, but after 22 it goes 0-8-16-0-8-16-0-8-16-0 

**To Reproduce**
Steps to reproduce the behavior:
1. Use the dayahead on a tag
2. Wait until you have the price for two days, instead of hours in the bottom it just repeats 0-8-16-0

**Expected behavior**
A few days ago this was working as intended.

**Screenshots**

[image]

nlimper, 2025-02-22:
Those times are returned by the api call to ENTSOE. They regularly have disruptions, and also some countries are switching to 15 minute timeframes instead of hourly, but I don’t know how that change will look. 
See https://newtransparency.entsoe.eu/news?id=67b72ba9fc0da26e7b5fbce4 for the status page. 
The problem will resolve by itself most of the time. If not, let me know, so we can reopen this issue and I will have a detailed look. In that case, also tell which country it involves.

dr3as, 2025-02-25:
This is still a issue. Country is Norway, zone is NO1.

nlimper, 2025-02-25:
The returned data is changed: it indicates an interval of 15 minute instead of 1 hour, but only every 4th value is returned so in effect it's still hourly... I will fix the google apps script to handle this. 

```
<Publication_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-3:publicationdocument:7:3">
<mRID>9d44250bd267482687c1c2fa82718017</mRID>
<revisionNumber>1</revisionNumber>
<type>A44</type>
<sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID>
<sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type>
<receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID>
<receiver_MarketParticipant.marketRole.type>A33</receiver_MarketParticipant.marketRole.type>
<createdDateTime>2025-02-25T10:07:33Z</createdDateTime>
<period.timeInterval>
<start>2025-02-24T23:00Z</start>
<end>2025-02-25T23:00Z</end>
</period.timeInterval>
<TimeSeries>
<mRID>1</mRID>
<auction.type>A01</auction.type>
<businessType>A62</businessType>
<in_Domain.mRID codingScheme="A01">10YNO-1--------2</in_Domain.mRID>
<out_Domain.mRID codingScheme="A01">10YNO-1--------2</out_Domain.mRID>
<contract_MarketAgreement.type>A01</contract_MarketAgreement.type>
<currency_Unit.name>EUR</currency_Unit.name>
<price_Measure_Unit.name>MWH</price_Measure_Unit.name>
<curveType>A03</curveType>
<Period>
<timeInterval>
<start>2025-02-24T23:00Z</start>
<end>2025-02-25T23:00Z</end>
</timeInterval>
<resolution>PT15M</resolution>
<Point>
<position>1</position>
<price.amount>38.83</price.amount>
</Point>
<Point>
<position>5</position>
<price.amount>42.99</price.amount>
</Point>
<Point>
<position>9</position>
<price.amount>39.36</price.amount>
</Point>
<Point>
<position>13</position>
<price.amount>36.94</price.amount>
</Point>
<Point>
<position>17</position>
<price.amount>35.16</price.amount>
</Point>
<Point>
<position>21</position>
<price.amount>46.76</price.amount>
</Point>
<Point>
<position>25</position>
<price.amount>49.29</price.amount>
</Point>
<Point>
<position>29</position>
<price.amount>51.45</price.amount>
</Point>
<Point>
<position>33</position>
<price.amount>52.68</price.amount>
</Point>
<Point>
<position>37</position>
<price.amount>53.72</price.amount>
</Point>
<Point>
<position>41</position>
<price.amount>53.33</price.amount>
</Point>
<Point>
<position>45</position>
<price.amount>52.37</price.amount>
</Point>
<Point>
<position>49</position>
<price.amount>51.75</price.amount>
</Point>
<Point>
<position>53</position>
<price.amount>51.51</price.amount>
</Point>
<Point>
<position>57</position>
<price.amount>51.63</price.amount>
</Point>
<Point>
<position>61</position>
<price.amount>52.35</price.amount>
</Point>
<Point>
<position>65</position>
<price.amount>55.81</price.amount>
</Point>
<Point>
<position>69</position>
<price.amount>55.7</price.amount>
</Point>
<Point>
<position>73</position>
<price.amount>55.57</price.amount>
</Point>
<Point>
<position>77</position>
<price.amount>53.13</price.amount>
</Point>
<Point>
<position>81</position>
<price.amount>53.7</price.amount>
</Point>
<Point>
<position>85</position>
<price.amount>51.69</price.amount>
</Point>
<Point>
<position>89</position>
<price.amount>50.12</price.amount>
</Point>
<Point>
<position>93</position>
<price.amount>49.9</price.amount>
</Point>
</Period>
</TimeSeries>
</Publication_MarketDocument>
```

nlimper, 2025-02-25:
google apps script script fixed, no firmware update needed.