OpenEPaperLink/OpenEPaperLink #444: Low on memory. Fallback to 8bpp

gitpower2017, 2025-02-28:
I have recently received this error message in the LOG.

16:02:29 File has size 0. /temp/FFFFFFFF1C01DC00.raw
16:02:28 low on memory. Fallback to 8bpp
16:02:28 Updating FFFFFFFFFF1C01DC00

The TAG does not accept any updates and the display is always the same. I have sent less information via JSON as a test. same result. 

Other tags are also affected.
Do I have to rewrite the file system on the ESP32?

The json string:

"mac=FFFFFFFF1C01DC00&json=[{"box": [1,21,169,3,3]},{"text":[5,0, "Müllkalender", "fonts/calibrib30", 1] },{"text":[2,30, "Graue Tonne", "fonts/calibrib30",1] },{"text":[171,30, "Heute", "fonts/bahnschrift30",1] },{"text":[2,55, "Gelber Sack", "fonts/calibrib30",1] },{"text":[171,55, "In 5 Tagen", "fonts/bahnschrift30",1] }]"

nlimper, 2025-02-28:
The AP is out of memory. Possible causes: the resolution of the tag is too big (but if a see the coordinates in the json content, it's just a small tag), or maybe there are several tags having pending data (are you updating multiple tags at a time?).

What kind of AP do you use?
Can you copy/past the status bar of the web interface? (`free heap: 184.03 kB ┇ free PSRAM: 7.63 MB ┇ db size: 5.71 kB ┇ db record count: 29 ┇ filesystem free: 5.37 MB ┇ uptime: 12h 35m 54s`)

gitpower2017, 2025-02-28:
It is only updated one day at a time. 

current env: OpenEPaperLink_Mini_AP_v4
build date: 2024-12-08 21:06
esp32 version: 2.75
filesystem version: 2.75
psram size: 8382631
flash size: 16777216
--------------------------

However, the oepAP has rebooted.
free heap: 176.30 kB ┇ free PSRAM: 7.96 MB ┇ db size: 2.72 kB ┇ db record count: 17 ┇ filesystem free: 4.53 MB ┇ uptime: 42m 9s

nlimper, 2025-02-28:
it's not about that particular tag, but about all tags connected to the AP: are multiple updated at once?
Anyway: does it work again now the AP has rebooted? Or does it never work with that tag?

gitpower2017, 2025-02-28:
I'm not 100% sure, usually only one is updated at a time and not all at the same time.
After the restart all tags work again. 
I have now activated the daily reboot.

nlimper, 2025-02-28:
In that case it is to be expected, due to memory fragmentation, the daily reboot is useful to enable (as it is by default).

closed 2025-02-28 by nlimper