OpenEPaperLink/OpenEPaperLink #445: Feature request: Dayahead prices option to show price unit

MaZZly, 2025-03-02:
**Describe the solution you'd like**

I would like to have the option to show the price unit in the "current price", e.g. `0.15c/kWh`

When enabled, the unit could be automatically set depending on the `Country` & `Price units`.

However, an easier option (implementation-wise) is to allow specifying whatever unit in the tag settings and just use that (trusting that the end-user enters it correctly)

nlimper, 2025-03-02:
I reather have no units, than the wrong units... 0.15c/kWh ? You probably mean 15c/kWh or 0.15€/kWh ;-)
But, not planned to add it, as it doesn't really add functionality. Of course, feel free to add it yourself. :-)

closed 2025-03-02 by nlimper

MaZZly, 2025-03-02:
In Finland (and many other EU countries) it is not uncommon to have prices as low as 0.1c/kWh.

We even have **negative prices** every now and then (more often in the summers, but also happens in the winters)

For example today, currently, the price is 0.25c/kWh:

[image]

https://oomi.fi/en/electricity/electricity-contracts/active/spot-price-of-electricity/

> But, not planned to add it, as it doesn't really add functionality. 

Well for family members who don't know the units, it really would make sense to know if it is cents or €

> Of course, feel free to add it yourself. :-)

Can you point me in the direction to which file this is generated, and I might take a crack at it later :)

nlimper, 2025-03-02:
That's very cheap! :-) In the Netherlands, prices are zero or negative in exceptional situations, but if you use display cents per kWh, there is no confusion possible as luckily it's never more then an euro per kWh. :-)

To add a parameter to the options of a content type: https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/ESP32_AP-Flasher/wwwroot/content_cards.json
The actual generation of the graphics: https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/ESP32_AP-Flasher/src/contentmanager.cpp#L1575-L1719

MaZZly, 2025-03-02:
> That's very cheap! :-)

Yep, tonight (with cheaper tariffs) the car gets charged 🤤 

> To add a parameter to the options of a content type: ...

The JSON looks straightforward, and although C++ isn't my forte, it doesn't look like it would be too dangerous to inject an optionally defined unit there 😅 

Are there instructions somewhere on how to run my own fork/version of the code on my AP? 🤔

nlimper, 2025-03-02:
For running your own fork: you can just use vscode/platformio to compile and send it via usb cable to your AP. Or, if you want to use OTA with your own fork, here is a description: https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Using-your-own-fork-with-OTA-updates