OpenEPaperLink/OpenEPaperLink #449: Problems with connection to Tag Flasher

EndOfYourGame, 2025-03-19:
Hi,

I'm trying to flash some tags using ESP_S2_mini flasher. Unfortunately something is wrong and I ran into a number of problems. First of all, I flashed the esp32 as shown here: https://github.com/OpenEPaperLink/OpenEPaperLink/tree/master/Tag_Flasher and it seems like everything is in order. I get no errors and I get terminal message that installation was successful. After that I tried connecting the tag to the Esp, also as shown in mentioned link, but I wasn't able to flash it. This is the message I get when I try to use OEPL-Flasher.py:

PS C:\Users\z0177476\Desktop\OpenEPaperLink-master\Tag_Flasher> python OEPL-Flasher.py -e -n -p COM10 read dump.bin -f --pt
Successfully opened serial port COM10
Connection with the flasher established. Version: 50
Failed to establish a connection to the microcontroller

I know my description might not be most detailed but I'm not sure how to even debug this situation, so I would appreciate any help.

nlimper, 2025-03-19:
`Connection with the flasher established. Version: 50` -> your flasher is working correctly.
`Failed to establish a connection to the microcontroller` -> your wiring between the flasher and the tag is incomplete or faulty. So check the pinout and the wires. Don't use breadboard jumper wires (they are famous for bad contacts), keep the wires short (about 10 cm). Post a clear picture of your setup if you cannot find the fault.

EndOfYourGame, 2025-03-19:
Yeah, I did use breadboard wires to connect. Thanks for clarifying what those messages meant, I'll make sure to make my connections better.

EndOfYourGame, 2025-03-24:
Hello again,

I have redone the connections twice at this point. I used a new, more trustworthy wire and made sure my soldering was clean. I even tried connecting to a different epaper tag but I still encounter the same issue. Below you can see my connections to both tag and ESP32.

[image]
[image]
[image]

nlimper, 2025-03-24:
Wiring looks okay as far as I can see. Are your sure your tag is compatible? (and has a nRF52811 processor). There are sadly also versions which use a different MCU.

EndOfYourGame, 2025-03-24:
Well, at this point I'm not sure. I'm using EL022F3WRA tags. I also have EL016H3WRA, but haven't tried connecting with those yet.

nlimper, 2025-03-24:
apparently even tags with the same type number can have different MCU's.
But the EL022F3WRA, according to the FCC data, has a SiLabs MCU. 
https://fccid.io/2AFWN-EL022F3WRA/Internal-Photos/Internal-Photos-6265997

Somebody made a firmware for SiLabs, but I'm not sure about the status and which tags are compatible.
https://github.com/OpenEPaperLink/Tag_FW_EFR32xG22

closed 2025-12-06 by jonasniesner