OpenEPaperLink/OpenEPaperLink #452: NaN for the Day Ahead pricing

cthulu, 2025-04-04:
**Describe the bug**
Since last night the Day Ahead pricing is displaying NaN (I am using the NL prices). The graph looks okay, it's just the "current value" that's showing NaN.

**To Reproduce**
Steps to reproduce the behavior:
1. Use day ahead pricing tag
2. Ensure you are showing the current value as well
3. Select Netherlands as option
4. See NaN

**Screenshots**

[image]

nlimper, 2025-04-04:
It’s a disruption at the ESTSOE API. Not a bug at the OEPL side. In general, it fixes itself.
BTW, the comings months, they will switch to 15 minute intervals for the dayahead prices. When this happens, most probably I need to change some of the OEPL scripts, so that can also take a few days. They expect this in June.

cthulu, 2025-04-04:
Thanks for confirming, I thought it might be issue on their side (I just didn't know how to confirm it, as I am using Tibber at home and that has correct data).

I've read about the 15min tarrifs, glad you're already thinking about it.

cthulu, 2025-04-05:
Just a note, to prevent a bootloop this morning I had to disable the day
ahead tag. That resolved the bootloop. I will re-enable once the upstream
problem is resolved.

nlimper, 2025-04-05:
I just fixed the bootloop if no data is present (in next release).
Today I will take a closer look to see if ENTSOE or TENNET changed something in the prices api call.

cthulu, 2025-04-05:
Thanks a lot!

nlimper, 2025-04-05:
https://newtransparency.entsoe.eu/news

> Dear Transparency Platform users,
> 
> Please be informed that Energy Prices 12.1.D values are missing  for Areas BZN HU, GR, SI, FR, AT (Seq 1), CZ, SK, NL, BE, RO, HR, BG, DE-LU (Seq 1), PL in Transparency Platform since 03.04.2025. Data provider is informed.
> 
> Apologies for any disruption this may have caused.

It's a disruption that's out of our control.

closed 2025-04-05 by nlimper