OpenEPaperLink/OpenEPaperLink #458: CJK = Chinese-Japanese-Korean no displaying

lukedoomer, 2025-04-19:
**Describe the bug**
according to the tutorial [Uploading & Using Custom Fonts](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Uploading-&-Using-Custom-Fonts,-with-examples)
cannot make openepaperlink display any cjk character

**To Reproduce**
1. upload popular fonts to /fonts
`droid-sans-mono.ttf`
`NotoSansMono-Regular.ttf`
3. upload local json template to root fs
```
[
    {"text": [5,5,"ABCぁ あ ぃ い ぅ うXYZ","fonts/droidsansmono.ttf",1]},
    {"box": [10,30,20,20,1]},
    {"box": [35,30,20,20,2]},
    {"triangle": [60,30,60,50,80,40,1]},
    {"text": [5,80,"ABCァ ア ィ イ ゥXYZ","droidsansmono.ttf",1]},
    {"text": [5,95,"Plain text 7x14_tf","7x14_tf",2]},
    {"text": [5,110,"Plain text t0_14b_tf","t0_14b_tf",1]},
    {"text": [135,5,"ABCㄅ ㄆ ㄇ ㄈ ㄉ ㄊXYZ","fonts/notosansmono.ttf",2]},
    {"text": [215,5,"70","fonts/bahnschrift70",1]},
    {"text": [150,80,"50","fonts/calibrib50",2,0]},
    {"text": [205,60,"80","fonts/calibrib80",2]},
    {"text": [90,35,"ABC一 二 三 四 五XYZ","notosansmono.ttf",1]},
    {"line": [10,120,290,120,1]},
    {"line": [10,115,290,115,2]}
]
```
6. See error
no any alphabet/character with custom font

**Screenshots**
[image]

**Expected behavior**
display CJK character properly

nlimper, 2025-04-19:
The current Truetype library we use, sadly doesn't do Unicode.
Two possible solutions: 
- add Unicode decoding to the current library (/src/truetype.cpp, function textDraw and/or codeToGlyphId)
- or, change to a different truetype render library. Here is a good one: https://github.com/takkaO/OpenFontRender , with the big advantage that it also parses the hinting tables.
Both options take quite a bit of work to implement (and test). If somebody has the time to spend on this, be my guest... I'm willing to take a look at it, but it will only be after the summer.

lukedoomer, 2025-04-19:
Thanks. Hope the unicode render will come to this great project.

nlimper, 2026-10-14:
The library did decode UTF-8 already, but it read the character widths from the wrong place for most CJK fonts, so characters overlapped and lines broke in odd places. That is fixed in commit f76323a7, which is in the latest release.

A few things to watch out for: the font name in the json has to match the file name exactly (your example uses droidsansmono.ttf while the file is droid-sans-mono.ttf, which is why nothing was drawn at all). The font has to contain the characters (Droid Sans Mono and Noto Sans Mono don't have CJK), and it has to be a TrueType font with glyf outlines, not a CFF/.otf font. A subset of a CJK font with only the characters you need saves a lot of flash. Feel free to reopen if it still doesn't work.

closed 2026-10-14 by nlimper