OpenEPaperLink/OpenEPaperLink #459: Chroma29 eat batteries in 2-3 days

DunklerPhoenix, 2025-04-25:
Heho
I bought a lot of 75 chroma29 on ebay ([link](https://www.ebay.com/itm/314849563630) and flashed all of them with full v0010 firmware. They connect very quick and without any issues to the AP (lilygo) and show the informations I set up.

But now the problem is that they all eat the batteries (2xCR2450) like ice cream. They go down in 2-3 days. I thought at the beginning that the lot of batteries was bad, but even a new lot gets eaten up fast.

The update interval is configured to 10min and the tags get an update every 12-24h.

Could this be a bug or do I need to make special configurations on the AP? I can not really imagine that all 75+ tags are broken. (Flashing worked without reflash serial number, so eeprom should be fine)

Greetings GP

skiphansen, 2025-04-26:
Did you have to restore the SN on the tags when programmed them ?  If so did you put in the actual SN from the label?

The reason I ask is that the first digit after the letter indicates a hardware difference and if it's wrong (a 0 instead of a 1 or versa versa) that could be the problem.

I'll double check the current consumption when I get a chance.  I have a couple of Chroma29s that have lasted several months, but that's with weather updates ever 3 hours.

Increasing the update interval from 10 minutes to 1 hour couldn't hurt.

skiphansen, 2025-04-26:
I just tested a JA1xxxxxxxB tag with FW 10 and it seems fine.  If you have a good current meter you should measure about 1.2 microamps while it is idle between updates.

I'll test a JA0xxxxxxxB next.

DunklerPhoenix, 2025-04-26:
Thank you.

I looked all up and the serial numbers on the web interface and the EPD are identical for all (also for the ones with repaired SN). The only thing that is different is that the EPD end with a B in the SN and on web it shows an X instead.

For now I wait for the arrival of the new batteries because I found some dead batteries and some with undervoltage in my two 100pcs batches. I don't trust them.

Measurements (if I read them correctly):
- about 5mA for startup
- about 7mA while going to "wait for data" 
- up to 20mA while getting data from the AP 
- about 1 microamps while in standby. 

[image]
[image]

Just for fun I put the voltmeter on the batteries while they are inside the EPD and the voltage fluctuates greatly while the EPD is starting, communicating and downloading (between 3.05V and 2.6V). So I really think my batteries are trash xD

skiphansen, 2025-04-26:
Let us know if your next batch of batteries fixes the issue or not.  

I bought  bunch of Tags (not Chroma29s) that had CR2450s which seemed to be ok so I've been using them in my Chroma29s (I'm cheap!).  I haven't been keeping close track, but the tags have run for at least several months with the weather forecast content type updating every 3 hours.

Every Chroma29 that I've bought had VERY dead CR2450s ... like in absolutely ZERO volts.

The batteries in all of the Chroma74s I bought were at least usable.  Of course they might be a lot NEWER.

DunklerPhoenix, 2025-05-11:
sooo lil update

new batteries, nearly same problem.
after 1.5 weeks they are empty. what I saw was that the tags redraw pretty often even  if nothing changed.

DunklerPhoenix, 2025-05-12:
it depends. this HA Tag os showing my Home Status my other tests (without HA) used the hours counter

I need to look further in the part with the serial logging. because I dont have much material here. will report back

DunklerPhoenix, 2025-05-12:
Is it normal that it conacts the AP so often?

[openepaperlink.log](https://github.com/user-attachments/files/20173726/openepaperlink.log)

Settings
[image]

skiphansen, 2025-05-12:
It *IS* if you are watching the WEB page from a browser and "Shorten latency during config" is set to yes.  

I can attest that checking in every 40 seconds will kill a set of batteries in about a week.

However your screen shot shows that you have "Shorten latency during config" set to no so it shouldn't been doing that.

Try rebooting your AP and then check of the setting is still no.

The good news (I guess) is that it's not updating the screen every 40 seconds.

Skip

DunklerPhoenix, 2025-05-12:
I rebooted it and cleared my browser cache. It still shows no :)
I disabled it weeks ago because I use the Home Assistant Integration and you recommended it in your Wiki.
I let it run some time and report with new log back.
(Sent from Phone. See edit for serial log from now)

-------- Ursprüngliche Nachricht --------Am 12.05.25 21:52 schrieb Skip Hansen :
skiphansen left a comment (OpenEPaperLink/OpenEPaperLink#459)
It IS if you are watching the WEB page from a browser and "Shorten latency during config" is set to yes.
I can attest that checking in every 40 seconds will kill a set of batteries in about a week.
However your screen shot shows that you have "Shorten latency during config" set to no so it shouldn't been doing that.
Try rebooting your AP and then check of the setting is still no.
The good news (I guess) is that it's not updating the screen every 40 seconds.
Skip

[
{
"target": "https://github.com/OpenEPaperLink/OpenEPaperLink/issues/459#issuecomment-2873806416",
"url": "https://github.com/OpenEPaperLink/OpenEPaperLink/issues/459#issuecomment-2873806416",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

DunklerPhoenix, 2025-05-14:
> I can attest that checking in every 40 seconds will kill a set of batteries in about a week.

It really looks like the sleep mechsnism has a problem.  It uses often 40000ms but also 60000ms and 540000ms and this in quite random order 😵‍💫  

(see updated link in last post)

DunklerPhoenix, 2025-05-14:
> I can attest that checking in every 40 seconds will kill a set of batteries in about a week.

It really looks like the sleep mechsnism has a problem.  It uses often 40000ms but also 60000ms, 600000ms and 540000ms and this in quite random order 😵‍💫  

(see updated link in last post)

DunklerPhoenix, 2025-05-14:
15.05. 01:24 CET+2 [openepaperlink13.log](https://github.com/user-attachments/files/20217090/openepaperlink13.log)

skiphansen, 2025-05-15:
Here's the problem:
```
[2025-05-14 23:37:47.304] nextCheckInFromAP 10
[2025-05-14 23:37:47.305] getNextSleep: 45 45
[2025-05-14 23:37:47.309] Sleeping for 600000 ms
[2025-05-14 23:47:47.394] No update
[2025-05-14 23:47:47.394] nextCheckInFromAP 0
[2025-05-14 23:47:47.397] getNextSleep: 45 45
[2025-05-14 23:47:47.398] getNextSleep: 45 45
[2025-05-14 23:47:47.403] Sleeping for 45000 ms
[2025-05-14 23:48:32.533] No update
[2025-05-14 23:48:32.534] nextCheckInFromAP 0
[2025-05-14 23:48:32.537] getNextSleep: 45 45
[2025-05-14 23:48:32.538] getNextSleep: 45 45
[2025-05-14 23:48:32.543] Sleeping for 45000 ms
[2025-05-14 23:49:17.654] No update
```
The AP told the tag to sleep for 10 minutes @ 23:37:47.304 which is correct based on your configuration.
The tag then sleep from 23:37:47.309 to 23:47:47.394 which is correct.
Then the AP didn't provide a sleep time with the next AvailDataInfo packet.

```c
struct AvailDataInfo {
    uint8_t checksum;
    uint64_t dataVer;  // MD5 of potential traffic
    uint32_t dataSize;
    uint8_t dataType;
    uint8_t dataTypeArgument;  // extra specification or instruction for the tag (LUT to be used for drawing image)
    uint16_t nextCheckIn;      // when should the tag check-in again? Measured in minutes
} __packed;
```
This caused the tag to use its internal power savings algorithm to calculate the sleep time and the result was 45 seconds.
From 23:48:32.534 to 23:57:19.134 the tag polled the AP every 45 seconds or so and the AP kept relying with zero for nextCheckIn.

At 23:57:19.134 the AP started to tell the tag to sleep for 10 minutes again.

At 00:36:59.696 the first new screen update became available.

This appears to be a AP bug to me.

What was the content type for this tag during this test?

If it wasn't a simple AP generated image please rerun the test with the current weather content with updates set once an hour.

There were changes to the update scheduling code in the 2.81 release that I am not familiar with, perhaps @nlimper can give us his take when he's back home from his trip.

The applicable tag code fragment is [here]( https://github.com/skiphansen/Tag_FW_Chroma/blob/1a2e2d15c3e805e634af6add0324f04b54cb3afb/Chroma_Tag_FW/OEPL/main.c#L140-L164) for review.

DunklerPhoenix, 2025-05-15:
Evil AP xD but an AP bug would be great. I dont want to reflash all tags :3

The tag is already set to current weather one hour for all the serial logs.

PS Is it possible to support your work with money?

nlimper, 2025-05-15:
As seen from the AP-side, this is all by design.
The OEPL protocol defaults to checkins around 40 seconds, on which a tag is supposed to last for 5 years on a set of batteries. At least, for the Solum tags. On the Chroma tags, in theory it should be possible too, as the original protocol of the manufacturer will do something similar.

For extra savings on top of that, there is the TTL parameter, to tell the AP that you don't expect to send a new image for x seconds. The AP lets the tag sleep for that amount of time (or shorter, as stated by the 'max sleep' setting in the config). After the TTL expires (that is, after the 'next update' time, as shown in the web interface), a new image is expected to be generated in any moment, so from then on, the AP lets the tag sleep at a default 40 second time. It's not wise to let the tag sleep for a long timespan again, as it would make the system very unresponsive.

For the build in content, this is working smoothly. But if you use HA to send images, and want to use the least amount of checkins, it's important to use the TTL parameter when you send the image, calculated as the time that you're actually plan to send the next image.

DunklerPhoenix, 2025-05-15:
what O.o 5 years? I would be happy if they last 3 months

skiphansen, 2025-05-15:
> least, for the Solum tags. On the Chroma tags, in theory it should be possible too, as the original protocol of the manufacturer will do something similar.

One very significant difference is that factory protocol for the Chroma tags is that the AP polls the tags for data.  It basically transmits constantly and the tags wake up periodically to **LISTEN** so they do **NOT** transmit periodically.  It takes **significantly** less current to receive than it does to transmit.

I don't want to engage you in a deep technical discussion while you're out enjoying nature! Are you back from your trip?

skiphansen, 2025-05-15:
> PS Is it possible to support your work with money?

Thanks for asking!

This is just a retirement hobby for me, using the results of my efforts (and helping me make it better) is all the thanks I need.

nlimper, 2025-05-15:
> the ap just broadcasts for eg the serial number of a tag when an update is available and the tag is just listening for his own serial number and then starts to update like it would do with the OEPL protocol?

It's *much* more complicated than you think. You can not just have the tag listening continuously, as even just listening already takes a lot of energy. So you would need some synchronised listening, with all risks on mis-timings etc, because the tags don't have stable timings.

DunklerPhoenix, 2025-05-15:
Yeh that was what I meant with I dont know anything about.
I had in my head:
AP always repeatly transmit a package whitch has just the information with tags to update. Eg each tag gets in the beginning an internal id and the AP just sends something like [1,5,6,7].

Sometimes a tag awakes after a hardcoded time and listens shortly for this and if it hear his id it beginns to communicate normaly.

AP transmit every 5ms, Tag listens for 25ms 
If no catch. Sleep and try again later.

That was just my limited simple thinking to this ^^‘

skiphansen, 2025-05-16:
> Aha! I didn't know that. So, using the current OEPL protocol, the Chroma tags can never be made as energy efficient as the Solum tags. :-( Probably the best thing to do, is to increase the default checkin time in the tag firmware, maybe to even something like 5 minutes (in that case, a simple addition to the AP will make sure it doesn't show the tag as timed out). They will be a lot less responsive, but at least the batteries will last. Or, do you have other ideas on saving power?

Just as a test I can certainly make a one off version for DunkerPhoenix prevents the tag from waking up every 40 seconds just to see if it fixes the problem for him.  I'll do that first thing tomorrow.

My feeling is that normally the tags haven't been checking in every 40 seconds, that's certainly not what I've seen.

I updated to 2.81 when you released it and I've been running it since then with a couple of Chroma29s here and I haven't changed any batteries yet.

I'm personally saw > 3 months with Chroma42s and weather forecasts every 3 hours with 2.75.  

3 months was just how long I ran them at our summer place before we hid out of the rest of the year.  After 3 months the batteries still tested like new.

I did kill a set of batteries in a week with checkins every 40 seconds because I had left my PC parked on the AP page while testing with the Shorten latency configuration option enabled.

> btw: yes, I'm back home! :-)

Welcome home!

skiphansen, 2025-05-16:
Here's a new version to try: [chroma29_full_0013_459_debug1.zip](https://github.com/user-attachments/files/20245173/chroma29_full_0013_459_debug1.zip)

Changes: 
1. When the AP replies with zero for nextCheckIn it will be replaced with 10 (minutes).

skiphansen, 2025-05-16:
@nlimper I still think that there's something odd happening on the AP side.

For example my test tag has been off since Wednesday and I just flashed it with the above firmware and started it up again.
The AP shows:

[image]

[image]

so Thursday and Friday were rendered at midnight as expected, but the AP is still relying 
No update.

It seems to me that since two updates are available it should be downloading them.

I'll keep logging until it updates, here's the log so far: 

[sh_5_16_25.log](https://github.com/user-attachments/files/20247438/sh_5_16_25.log)

I'm going to study the scheduling code in the AP when I get time to get a better understanding.

nlimper, 2025-05-16:
That can happen if the S3 and C6 get out of sync: the S3 told the C6 that there is an update available, but the C6 has since then reset for some reason, so it doesn't tell the tag that there is an update. Right click the tag and delete the pending image, and then force update to generate a new image.

skiphansen, 2025-05-16:
Yes, that fixed it, the tag updated to the most recently generated image.  Now to figure out why the C6 reset!

However the older images weren't deleted.  Should we ever had more than one pending image for a particular tag?
I have nightly reboots disabled just to check stability, to we rely on the reboot to clean up old images?

nlimper, 2025-05-16:
Clean up takes place on the 'transfer complete' signal, and at restart. I guess 'delete pending image' option doesn't remove it from the flash.
I'm sure there are thousands of little improvements that can be made, but please for now lets focus on the original problem. :-)

DunklerPhoenix, 2025-05-16:
16.05. 23:18 CET+2 (gets updated now and then)

[openepaperlink13.1.log](https://github.com/user-attachments/files/20259298/openepaperlink13.1.log)

It froze on first boot
```
[2025-05-16 22:53:42.646] Sleeping for 10000 ms
[2025-05-16 22:53:52.680] No saved channel
[2025-05-16 23:05:13.819]   <- manual reboot
```

It is good that the "nextCheckInFromAP 0" still is 40s on blk fail
```
[2025-05-16 23:07:16.486] blk failed validation!
[2025-05-16 23:07:16.487] nextCheckInFromAP 0
[2025-05-16 23:07:16.491] getNextSleep: 40 40
[2025-05-16 23:07:16.495] getNextSleep: 40 40
[2025-05-16 23:07:16.495] Sleeping for 40000 ms

```

skiphansen, 2025-05-17:
> 17.05. 00:49 CET+2 (gets updated now and then)
> 
> [openepaperlink13.1.log](https://github.com/user-attachments/files/20259819/openepaperlink13.1.log)
> 
> It froze on first boot

There's probably the [known issue](https://github.com/OpenEPaperLink/Tag_FW_Chroma/issues/2) with the first sleep at power up.  I was never able to find it.  It's very difficult to repoduce.

> 
> ```
> [2025-05-16 22:53:42.646] Sleeping for 10000 ms
> [2025-05-16 22:53:52.680] No saved channel
> [2025-05-16 23:05:13.819]   <- manual reboot
> ```
> 
> It is good that the "nextCheckInFromAP 0" still is 40s on blk fail

Right, nextCheckInFromAP  is only overridden when the AP replies "no update" with nextCheckInFromAP of zero,
in other cases the nextCheckInFromAP is used as is.

Fingers crossed...

skiphansen, 2025-05-17:
I think the C6 reset issue **IS** the causing this issue.

I configured the content for my test tag as current weather with 30 minute updates and ran it all day.  It appears that the C6 reset twice which caused an update to be lost which invoked my kludge which forced a 10 minute sleep.

```
9:27:02.452: Drawing image in slot 0
9:49:13.859: Drawing image in slot 1
10:19:26.840: Drawing image in slot 2
10:49:38.302: Drawing image in slot 3
11:19:51.322: Drawing image in slot 0
11:50:02.723: Drawing image in slot 1
12:19:15.694: Drawing image in slot 2
12:49:28.694: Drawing image in slot 3
13:19:41.666: Drawing image in slot 0
13:49:54.707: Drawing image in slot 1
14:19:03.119: Forced nextCheckInFromAP to 10 minutes
14:29:03.206: Forced nextCheckInFromAP to 10 minutes
14:39:03.284: Forced nextCheckInFromAP to 10 minutes
14:49:07.910: Drawing image in slot 2
15:12:20.896: Drawing image in slot 3
15:42:33.915: Drawing image in slot 0
16:12:46.868: Drawing image in slot 1
16:41:55.350: Forced nextCheckInFromAP to 10 minutes
16:51:55.438: Forced nextCheckInFromAP to 10 minutes
17:01:55.474: Forced nextCheckInFromAP to 10 minutes
17:12:00.118: Drawing image in slot 2
17:35:13.134: Drawing image in slot 3
18:05:26.155: Drawing image in slot 0
18:35:39.159: Drawing image in slot 1
```

So normally the tag sleeps until the next update is available, but without the kludge the tag polls the AP every 40 seconds until a new update is available when the AP replies w/o an update with a nextCheckin of 0.

I'll add AP logging to my test and see what I can find.

skiphansen, 2025-05-19:
I don't know much about HA, but I won't think so.  The preview on the AP normally updates after the tag updates the image and acknowledges the transfer.

skiphansen, 2025-05-20:
@nlimper It doesn't look like the C6 is resetting.  

The problem appears to be that occasionally updates are not generated, but expectedNextCheckin is advanced anyway.

I haven't been able to trace down the cause of this yet.

My latest logs are attached if you want to take a look.

The MAC of my test tag is 4467090000013753.  The problem starts @ 7:15:14 after a successful update.  Since the content type is current weather with 30 minute updates the next update should be around 7:45, but the update isn't generated.

The AP code is the current master as Sunday of plus debug logging.  The code is committed here: https://github.com/skiphansen/OpenEPaperLink/tree/debug_issue_459

Ap log:  [5_20_25_ap.log](https://github.com/user-attachments/files/20350836/5_20_25_ap.log)

Tag log: [5_20_25_tag.log](https://github.com/user-attachments/files/20350837/5_20_25_tag.log)

nlimper, 2025-05-21:
> The problem appears to be that occasionally updates are not generated, but expectedNextCheckin is advanced anyway.

> The MAC of my test tag is 4467090000013753. The problem starts @ 7:15:14 after a successful update. Since the content type is current weather with 30 minute updates the next update should be around 7:45, but the update isn't generated.

At 7:13 everything runs according to plan: a new image is generated, the tag picks up the new image, and the nextupdate is set to 07:43:57 and the tag is set to sleep for 28 minutes.

But then, At 7:43, a new image should be generated, but for some reason expectedNextCheckin was set to 8:11, which is too far ahead in the feature so the generating of the image is delayed until 5 minutes before 8:11 (to prevent an image to be generated with data that is outdated when it finally gets displayed).

I don't know yet why/how expectedNextCheckin is set to 8:11. I will investigate.

DunklerPhoenix, 2025-05-22:
> [@DunklerPhoenix](https://github.com/DunklerPhoenix) I'm still interested in knowing if the debug firmware resolves your problem w/o an AP change.
> 
> I'd suggest putting in a new set of batteries on one of your TAGs and then running an endurance test with your intended use case.
> 
> Do **NOT** leave a serial port or programmer connected during the test. These connections can effect the true battery drain.

Sooo the test is set up.

Tags: 
1x - the original tag with the debug software that is already running some time with disconnected serial
3x - Home Assistant provided updates (home/away) with fresh batteries of the same lot
3x - hourly count up provided by AP with fresh batteries of the same lot

DunklerPhoenix, 2025-05-22:
nope
by hand like all others xD
thank you for the ota. i'll use it for the other 10 that are active atm

DunklerPhoenix, 2025-05-30:
The tags that count hours are mostly dead now. 

1x - the original tag that is already running some time with disconnected serial - dead
3x - Home Assistant provided updates (home/away) * - all alive
3x - hourly count up provided by AP * - 2 of 3 dead

skiphansen, 2025-05-30:
Thanks for all of the effort you've put into this and the feedback!!!

I'll do some measurements and do some calculations and see how may hours it should last in theory. 

The manufacture's spec for battery life of the Chroma29 is 5 years with 3 updates per day.
Assuming updates use most of the battery that would still be 7 1/2 months with hourly updates.

Of course the factory protocol didn't require the tag to transmit as often as the OEPL protocol so there's that.

BTW it's better not to edit comments days later ... edits don't send notifications to people watching this issue, but new comments do.

DunklerPhoenix, 2025-05-30:
thank you
I know. The edits were made on purpose to prevent that many emails. 

If you want I can send you as much of my tags as you like via post, if you need them.

skiphansen, 2025-05-30:
Thanks for the offer, but I'm good.  I bought the same quantity from the same guy and have lots that I've never programmed.
I'm am interest in any other SubGhz tags that I don't have if you (or any lurkers) happen by any.

DunklerPhoenix, 2025-06-22:
Update: I let all tags (75+) running updates of the current hour. Only 4 Tags are still running and are over 1200h+. The batteries are still pretty full and they don't use the debug software with the minimum time. 

SOOOOO all other Tags of that lot are broken and only this 4 are usable. Maybe I didn't have any luck with it or I can't recommend buying them from that ebay seller.

skiphansen, 2025-06-30:
> Update: I let all tags (75+) running updates of the current hour. Only 4 Tags are still running and are over 1200h+. The batteries are still pretty full and they don't use the debug software with the minimum time.
> 
> SOOOOO all other Tags of that lot are broken and only this 4 are usable. Maybe I didn't have any luck with it or I can't recommend buying them from that ebay seller.
> 
> Edit: The 4 working one are JA100XXXXXB Tags and not the old version

Are there any JA100XXXXXB tags in the "broken" batch?

I find it hard to believe the tags are actually "broken", there's just not much to go wrong which would only effect battery life.

A more likely answer Is that the firmware isn't going into the low power sleep mode quite correctly and minor differences between boards are begin magnified.

If you haven't tossed them you I'd like a couple examples of the tags that preformed the poorest in your testing.

DunklerPhoenix, 2025-07-02:
I have 4 addional JA1. They hold between 200 to 900h before they are dead.
The JA0 tags only come to max 250h before they die.

I don't throw them away. Even if they don't work on battery I can still use them on another power source.

I can't find any option to write a DM, but this is a throw away mailadress. I will remove it after our exchange.
openepaperlink.sfrsu@8alias.com

skiphansen, 2025-07-16:
There's a problem alright! Your JA003xxxxxB tag measures 2.93 milliamps in sleep mode.  

My development Chroma29 S/N JA000xxxxxB measures 1.5 microamps in sleep mode.

None of my Chroma29s that I bought ordinally got from the same vendor included any
JA003xxxxxB tags, they were all either JA000xxxxxB or JA100xxxxxB.

So my current guess is that there are hardware difference between JA000xxxxxB
and JA003xxxxxB that I'm not accounting for in the code.

Unfortunately I see that there are also JA001xxxxxB and JA004xxxxxB Chroma29s exist.

Isn't reverse engineering fun?!?

I'll take some tags apart and see if I can figure out the differences.

Thanks again for helping the project!  From the packing material you use
I can tell you spend some money on batteries!

skiphansen, 2025-07-16:
It looks like an easy fix!

The pcb on JA003xxxxxB boards matches the one used on JA1xxxxxxxB board so I just need to modify the code that parses the SN to determine which of the two power saving configurations to use.

I should have a new image for you to test in the next couple of days.

DunklerPhoenix, 2025-07-18:
JA0008XXXXX 3,144V
JA0016XXXXX 3,184V 
JA0020XXXXX 3,180V
JA0036XXXXX 3,120V
JA1001XXXXX 3,208V
JA1002XXXXX 3,204V

All with FW21 0x15 on channel 104.
Now counting hours with max sleep 300. (I need 300s for the other tags that are in production use at the moment)

skiphansen, 2025-07-20:
Looking good!

[image]

[image]

DunklerPhoenix, 2025-07-20:
totally forgot I can look at the graph from HA xD

Here is the curve from the "strange" JA0008 Tag

[image]

The graphs of the other tags look cpmplety different
here as an example the JA0020

[image]

DunklerPhoenix, 2025-07-20:
nevermind... if i throw them side by side the look quite similar
maybe the battery had a problem at start or it measures the voltage wrong

[image]

skiphansen, 2025-07-29:
The big drop on 21 Juli looks suspicious, any idea what caused that?

My test using your old 20 hour tag is still looking good. I started the test with a fresh set of year old tenergy CR2450 batteries.

[image]

[image]

DunklerPhoenix, 2025-07-29:
nice to hear.
nope I dont know whats happened there. I didn't touch the tag in that time, but we will see in the second round if I can repeat it :3

skiphansen, 2025-07-30:
> Off-Topic Update
> 
> JA0036XXXXX 3,120V This one froze 50 hours ago. It's still communicating with OEL, but OEL doesn't seem to update it anymore.
> 
> Somehow (idk why) it switched from hour counting to remote content. It was integrated into HA some weeks ago. So maybe the HA integration did something strange.
> 
> <img alt="Image" width="2000" height="563" src="https://private-user-images.githubusercontent.com/1261305/472540220-8adc6dd2-a71b-46d6-b9f1-36f8ac2dba69.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM5MDg3MTIsIm5iZiI6MTc1MzkwODQxMiwicGF0aCI6Ii8xMjYxMzA1LzQ3MjU0MDIyMC04YWRjNmRkMi1hNzFiLTQ2ZDYtYjlmMS0zNmY4YWMyZGJhNjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDczMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MzBUMjA0NjUyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzJjZmYyMmRlYTJlZjI4ZTdkMTY5ZDc0ZDgyNzUwMzEzZmIyZTc2MzlmMTYzZDkyZmYxNTU5MjgwMzIxYjlmZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.3xO08__ForFS18IowTwBy0rthcv5YJ0ziuwIp93bmTE">

Sorry I have no idea, but I'm 99.9% sure it's not related to the tag's firmware.

DunklerPhoenix, 2025-08-09:
JA0008XXXXX (restarted 2 weeks ago 3,12V) -> 2,32V
Still working, but reporting low battery
Hours count 267
(other testtags 530h+ and still over 3,0V+)

It seems that this one still has a problem. If its empty I will open it and post images of the board here. 
There is again a strange voltage drop in the chart like on the first cycle. (nobody touched it)

[image]

skiphansen, 2025-08-31:
[image]

My test has broken the 1000 hour threshold and the battery voltage is still above 3 volts. 

I'm closing this issue as fixed.

closed 2025-08-31 by skiphansen

skiphansen, 2026-01-20:
[image]

Battery voltage still looking good, but I'm not sure how accurate the measurement is.

skiphansen, 2026-09-17:
Shocking enough still running here!

[image]

So over a year and still ticking (barely .. down to 2.28 V it's won't be long now.

DunklerPhoenix, 2026-09-17:
10k 10k 10k ✨

skiphansen, 2026-10-23:
Well I must say I'm quite surprised !
I've been on vacation and I expected the miss the big event but ... it's still running !

[image]

skiphansen, 2026-11-11:
"He's dead Jim!"

[image]

Display showed 10964 updates when it died.

DunklerPhoenix, 2026-11-11:
10964 hours
456.8 days
65.2 weeks
14.7 months
1.2 years
🫡 salut for a hero

skiphansen, 2026-11-11:
And now I have reasonable data points to make a reasonable guess about percentage of battery left.  At least for this tag.