OpenEPaperLink/OpenEPaperLink #462: Sometimes double refresh on Solum M3 2.9" Lite

olanwe, 2025-05-02:
**Describe the bug**
Certain updates cause the tag (Solum M3 2.9" Lite - EL029H4WRC, Tag FW 0029-ARIA) to get refreshed twice. Not sure if this is a problem related to the tag FW or the AP (running FW 2.81), because it is sometimes difficult to reproduce. 

**To Reproduce**

This works on my side - at least today...
`[
  {"image": ["images/no_stream_128x102.jpg", 33, 33] },
  {"textbox": [174, 62, 190, 60, "Der Störungsmelder\nist nicht erreichbar!", "bahnschrift20.vlw", 1, 1.25] }
]`
Steps to reproduce the behavior:
1. Upload image file to images/ folder
2. Open JSON-upload form
3. Insert JSON:
`[ {"image": ["images/no_stream_128x102.jpg", 33, 33] }, {"textbox": [174, 62, 190, 60, "Der Störungsmelder\nist nicht erreichbar!", "bahnschrift20.vlw", 1, 1.25] } ]`
and upload it to the tag
4. Check tag and log

It helps playing around with the content if it does not show this glitch at first.

**Expected behavior**
Log should show:
```
    13:06:14 xxx reports xfer complete
    13:06:14 xxx block request /current/xxx_865804.pending block 0, len 1043 checksum 750
    13:05:15 new image: /current/xxx_865804.pending
    13:05:13 Updating xxx
```

The tag is then updating the display once.

**Observed behavior**
Instead, the block request is not being logged - although the change is actually being transmitted and the intended upload must have taken place. The tag refreshes the display twice: The first time right after upload and the second time on the following data poll.

```
    13:08:16 xxx reports xfer complete
    13:07:15 xxx reports xfer complete
    13:06:45 new image: /current/xxx_956263.pending
    13:06:43 Updating xxx
```

**Additional context**
I have also tried several variations to find the origin of the problem:
- Send this JSON via upload form to the tag: glitches
- Send the exact same JSON to another tag type (e.g. 4.3" with same firmware) without any modification: works!
- Upload jpeg-image of generated JSON: glitches
- Just change text (e.g. replace a character): glitches
- Remove image and only leave text: works
- use different font: works
- tried older tag firmware 0029-BUGF: glitches too.

Image used in the test
[image]

Rendered jpeg-image that glitched when being uploaded onto my tag (using a different font here):
[image]

RalfEs73, 2026-09-23:
I'm seeing what looks like the same issue (double "xfer complete"), but in a more severe form on a large M3 tag: the new image is never displayed.

**Hardware / versions**
- Tag: Solum M3 12.2" (960x768, BWR), tag FW `0x46`
- AP FW: 2.92
- Content is pushed from Home Assistant via the OpenEPaperLink integration

**Radio / power look fine**
- Battery 2.98 V (fresh batteries), RSSI -57, LQI 216, check-in interval ~60 s

**Log when it fails**

`22:09:28 Updating 980C33FFFE7183C7
22:09:33 new image: /current/980C33FFFE7183C7_557708.pending
22:09:49 980C33FFFE7183C7 reports xfer complete
22:10:29 980C33FFFE7183C7 reports xfer complete`

(Same pattern earlier at 20:35/20:36 and 21:02/21:03.)

**Behavior**
- The tag downloads the image completely, then downloads it again on the next poll ("xfer complete" twice), but the e-paper is **not refreshed** – the old content stays.
- The thumbnail in the AP web UI briefly shows the new image and then switches back to the old one after a few seconds.
- While the tag is in this state, even a minimal image (a single text element "Test") is not displayed, so it does not seem to depend on the content/size of the image.

Happy to provide more logs or test builds if that helps.

nlimper, 2026-10-14:
The latest release fixes part of this. On a repeated "xfer complete" the AP set the tag's image hash to random memory, so after the second xfer complete it thought the tag showed a different image and sent the same content again (commit 84adea3b). The latest release also has a number of radio chain fixes in the C6 firmware that could explain why the tag reports the transfer twice, but I haven't been able to confirm that for this case. Could you update the AP and the C6 to the latest release and check whether you still see the double "xfer complete" in the log?

@RalfEs73 your case, where the image never shows up on the 12.2", may be something different. Please retest with the latest release too, and if it still happens, post the log lines of one failing update plus the serial output of the AP around that time.