OpenEPaperLink/OpenEPaperLink #464: Feature suggestion: Add gzip decompression for VLW-files

olanwe, 2025-05-13:
**Is your feature request related to a problem? Please describe.**
VLW font files are inefficient in terms of storage: they are uncompressed and waste one full byte per monochrome pixel! VLW-files can be significantly compressed, e.g. from 215 KB down to just 16 KB!

**Describe the solution you'd like**
Uploaded VLW files that have been gzipped before (.vlw.gz) could be automatically decompressed. This will save us a **lot** of flash-memory!

**Additional context**
I have already created two functions for testing purposes (see attachment) that will act as wrappers for the spr.loadFont() and spr.unloadFont() calls. They seem to work fine so far. However, I did not create a PR yet, because:
- I am neither an ESP32 nor an CPP expert, so I am not sure if the coding is OK (e.g. decompression takes place in memory - not sure if this might eat up too much RAM...?) 
- Where shall the functions be placed? (I tested them in contentmanager.cpp)

[decompress_gzvlw_fonts.txt](https://github.com/user-attachments/files/20194592/decompress_gzvlw_fonts.txt)

nlimper, 2025-05-14:
That could be a useful addition, thanks. I will integrate it at some point.

nlimper, 2026-10-14:
Thanks again for the idea and the code. Gzipped fonts are supported since 8eaf2661, based on your approach: upload name.vlw.gz instead of name.vlw, templates don't need to change. The default fonts are now shipped gzipped too, which saves about 230 KB. This is in the latest release. Note that an existing .vlw file takes precedence over the .vlw.gz, so delete the old .vlw files to actually free the space. Feel free to reopen if you run into problems.

closed 2026-10-14 by nlimper