OpenEPaperLink/OpenEPaperLink #478: Buitenradar http 404

sebnaf, 2025-06-24:
**Buienradar 404**
Selecting Buienradar returns a 404 from the service.

`18:39:52 Buitenradar http 404
18:39:51 get buienradar`

sebnaf, 2025-06-24:
Ein GET auf https://gadgets.buienradar.nl/data/raintext/?lat=48.7&lon=7.51 (Emmendingen in DE) liefert ein 404 "Not found: location must be inside the Netherlands or Belgium."

Der Service funktioniert also nur noch in NL oder BE.

Kostenlose Alternativen (in Reihenfolge meiner Präferenzen):

DWD Open Data (über Bright Sky API):
https://brightsky.dev/docs/#/operations/getWeather

Morgenwirdes.de
Beispiel: https://morgenwirdes.de/api/v2/rjson.php?plz=84051 die Postleitzahl auf 84051 gesetzt werden und für Koordinaten und Nutzungsbedingungen gilt das Gleiche wie oben. Die Daten sind Folgendermas aufgebaut. "time" beschreibt die Uhrzeit im Format hh:mm "timestamp" ist der Unix Timestamp des Datensatzes und "dbz" der Radar Reflektivität in dBz. Aktuell gehe ich davon aus, dass es ab ≈12.2dBz regnet.

jonasniesner, 2025-06-24:
If we want to use morgenwirdes for Germany, I could optimise the API

sebnaf, 2025-06-24:
Just sent you an email :) - short: yes