OpenEPaperLink/OpenEPaperLink #481: Feature request: Descriptions for Elements in Tagtype Template

MagnusErler, 2025-07-06:
**Is your feature request related to a problem? Please describe.**
Currently, the tagtype templates contain various numbered sections (e.g., "1", "2", "4", "16", etc.) with keys and data structures that are not self-explanatory. It is unclear what these numbered sections represent and what options or parameters are valid for each element. This makes it difficult to understand, edit, or extend the templates, especially for new developers or users unfamiliar with the format.

**Describe the solution you'd like**
I would like to have a description or schema for each element and section in the tagtype template. This could be:
- Inline comments
- A separate documentation file
- A formal JSON schema with field descriptions

For example:
- What does "1" represent? What options can be used within it?
- What are the possible keys (like weekday, date, location, wind, etc.) and what does each parameter in their arrays mean?

**Describe alternatives you've considered**
I’ve looked for existing documentation but haven’t found anything detailed for the specific element IDs and their possible configurations.

Reverse-engineering the structure by trial and error, but this is time-consuming.

atc1441, 2025-07-06:
Hey, 
The corresponding descriptions are here: https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/ESP32_AP-Flasher/src/contentmanager.cpp#L337

nlimper, 2025-07-06:
The id's in the tagtype json refer to the id's of the content cards (listed in https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/ESP32_AP-Flasher/wwwroot/content_cards.json)

The content within template for every id the is just a bunch of values which are use to draw the content in https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/ESP32_AP-Flasher/src/contentmanager.cpp

nlimper, 2025-07-06:
BTW, I'm not planning to write documentation for this, and I don't think anyone will, so I'm closing this. But of course feel free to add it to the wiki or write the documentation. :-)

closed 2025-07-06 by nlimper