OpenEPaperLink/OpenEPaperLink #484: Can you use an esp32-c3 with no display to make an Spaghetti AP?

ScotsFrog1, 2025-07-18:
Can you use an esp32-c3 with no display and a esp32-c6 to make an Spaghetti AP?, don't need a display, but the pinout on the non display version doesn't have pin 19 or 47, there is a RST, can this be done?

Thanks

ScotsFrog1, 2025-07-18:
> You can not use a C3 you will need an S3 with 16MB of flash and 8MB of ram

sorry is an s3

skiphansen, 2025-07-20:
> Can you use an esp32-s3 with no display and a esp32-c6 to make an Spaghetti AP?, don't need a display, but the pinout on the non display version doesn't have pin 19 or 47, there is a RST, can this be done?
> 
> Thanks

According to [this](https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Access-point-pinouts) page Pin 19 is debug-txd which isn't critical but is **VERY** handy when things don't work.  Like when your building AP from scratch.

Pin 47 is RST so you're good there.

If I were you I'd order the correct board they are cheap and might save you some grief.

The display isn't essential, but like the debug port it can be very handy.

closed 2025-07-22 by skiphansen