OpenEPaperLink/OpenEPaperLink #490: Corrupt camera entity pictures

emufan, 2025-08-06:
**Describe the bug**
I just updated to newest version of integration as well as to AP FW 2.81 and Tag FW 41 0x29

If I run now such an automation for devices, e.g.

```
      - action: open_epaper_link.drawcustom
        target:
          device_id:
            - a152781569ac5e7505baa098790c0af0
            - 3d2ca1c37f3314bd3cdc721168d1f6bf
        data:
          dry-run: false
          background: white
          rotate: 0
          payload:
            - type: text
              value: iX Batterie
              x: >-
                {% if states('binary_sensor.bmw_ix_weiss_ladezustand') == 'on'
                %}
                  165
                {% else %}
                  173
                {% endif %}
              "y": 48
              anchor: mm
              size: 28
              color: black
```

in generated the right image in the folder \config\www\open_epaper_link and send it to the tag.

[image]

BUT (never had that before the updates):

1. After the AP has sent it to the tag, the picture is deleted again in the folder.

2. If I have the camera entity on a dashboard or open it in more-info, the image is re-created but corrupt
[image]

**Expected behavior**
Have the last pictures still available for the camera entity.

atc1441, 2025-08-06:
Hey again, I did delete that comment again as Im not sure if it still counts as for me the Compressed images do show correct in HA,
In theory you could edit out the deletion of the images in the HA plugin and see if it works out. You can remove those 2 lines https://github.com/OpenEPaperLink/Home_Assistant_Integration/blob/main/custom_components/open_epaper_link/camera.py#L363 via an text editor in your HA installation and reboot HA

emufan, 2025-08-06:
Thanks to both of you. Don't know where 29 or 99 come from, but can test it.

But the approach in deleting the deletion (hint from above), but the request/update from AP as well (still searching for it, any tips) would be more what I would like.

atc1441, 2025-08-06:
29 is the minimum version from which G5 Compression is supported and "99" would be a very high number to "disable" it

"but the request/update from AP" I dont understand what you mean by this

emufan, 2025-08-06:
> "but the request/update from AP" I dont understand what you mean by this

I understood, that https://github.com/OpenEPaperLink/Home_Assistant_Integration/blob/main/custom_components/open_epaper_link/camera.py#L363 is just deleting the (working) image from folder. But thought the problem is the step afterwards, the pull of the compressed image from AP/tag if opening or using the camera entity. So I would think that I have to skip it as well, as the main problem.

emufan, 2025-08-06:
Great. Will do it that way and delete my camera snapshot way, which I used currently, again for now. Thanks so much for your help and the background.

g4bri3lDev, 2025-09-13:
With Release 2.0.0 of the Integration decoding of G5 images has now been implemented correctly so this issue should be fixed.

closed 2025-09-13 by g4bri3lDev