OpenEPaperLink/OpenEPaperLink #503: Epaperlink update to 2.82 does not update filesystem

jum0n, 2025-09-14:
**Describe the bug**
Filesystem shows 2.81 instead of 2.82 when doing full update and also if individual update

**To Reproduce**
Steps to reproduce the behavior:
1. Update and reboot access point
2. Check esp32 and filesystem version should match

**Expected behavior**
Filesystem should update to version 2.82

**Screenshots**
current env:        ESP32_S3_16_8_YELLOW_AP
build date:         2025-09-12 11:39
esp32 version:      2.82
filesystem version: 2.81
psram size:         8383655
flash size:         16777216
ESP32-C6 version:   0019
--------------------------

**Additional context**
Add any other context about the problem here.

kvandt, 2025-09-16:
Can confirm. Mini AP V4.

kvandt, 2025-09-17:
Yes, did multiple times. The ESP32 version was correct in the overview after updating. But strangely enough it is now correct after several power cycles of the AP. Must be some caching issue somewhere.

jum0n, 2025-10-03:
> But strangely enough it is now correct after several power cycles of the AP. Must be some caching issue somewhere.

Sorry for delayed reply but now that I check, mine displays the correct version as well with no action from me.  As stated, probably cached and then expired.