OpenEPaperLink/OpenEPaperLink #508: Feature request:

Hellis81, 2025-09-29:
**Is your feature request related to a problem? Please describe.**
Is there anyone who could add support for the 3.7 inch Gicisky display?

[image]

It's a very thin device with built in rechargeable battery, a button and I believe three pages you can cycle between using the button.
Home Assistant integration detects it as a 2.9 inch device and produces this kind of images 

[image]

I have also tried the ATC BLE page and it produces the same kind of images. 

**Describe the solution you'd like**
Support for it

**Describe alternatives you've considered**
I can help remotely but if that is not enough then I'm willing to buy one of these screens for anyone willing to get it working.

r2kch, 2025-10-03:
Nice display! If you are part of the discord channel, there is a section for the ATC BLE firmware. if not, its worth to join ;) 

As far as I remember, ATC1441 wants to have "hands on" for firmware support, not sure this is still the same.

skiphansen, 2025-11-04:
You can find the discord invite link at the top of [https://openepaperlink.de/](https://openepaperlink.de/), just click the Discord icon. (top right).